nxp-imx/mfgtools #177: booting a closed system with a CAAM device

ldts, 2020-03-31:
When booting a booting a closed system with a CAAM device (via uuu SPL for instance), SPL must be signed with

> [Authenticate CSF]
> [Unlock]
> Engine = CAAM
> Features = MID, RNG

See 
https://docs.foundries.io/latest/reference/secure-boot.html#booting-a-closed-system-with-a-caam-device

This works fine when the system boots from mmc but  when using the mfgtools the CAAM remains locked for changes and therefore panic on boot. Is there any option in the uuu boot command that would allow for those unlock instructions to be passed to the bootrom?

thanks

nxpfrankli, 2020-03-31:
which chip?

ldts, 2020-03-31:
I am testing on imx7ulp but I suspect the same will happen on any other chip, wont it?

nxpfrankli, 2020-03-31:
I am not familiar with this user case. 
I am not sure I understand correctly.   Signed SPL already boot from close chip?
you want a customer command to operator CAAM? does it just write a register?

ldts, 2020-03-31:
Yes SPL can boot with a signed image from a close chip. However, when the user needs to have access to the CAAM on a closed device (ie to reconfigure it), the Code Signing Tool provides the the UNLOCK command.

From the manual (CODE SIGNING TOOL, User's Guide, Rev 3.1.0, 09/2018)

> 
> The Unlock command prevents specified engine features from being locked when exiting the
> internal boot ROM. 
> Multiple Unlock commands may appear after the Authenticate CSF command. A feature will be
> unlocked if specified in one or more Unlock commands. Table 23 lists the Unlock command
> arguments.
>  
> example:
>  [Unlock]
>  Engine = CAAM
>  Features = RNG

When an imaged is signed this way, the CAAM can be initialized and configured. Otherwise, read/write accesses to some of the configuration registers will cause core aborts.

So, we can boot an image signed to support CAAM on a closed device when booting from MMC. However booting via mfgtools seem to ignore the provided UNLOCK command (ie, CAAM remains locked...somehow the bootrom is not doing the necessary setup).

ldts, 2020-04-01:
um...I cant figure out why this cmd encoded in the CSF region would not execute along the other commands

ldts, 2020-04-02:
is this the complete list of commands that the rom kernel understand? maybe there is another one to execute the CSF commands before jumping to the DCD?
#define ROM_KERNEL_CMD_RD_MEM							0x0101
#define ROM_KERNEL_CMD_WR_MEM							0x0202
#define ROM_KERNEL_CMD_WR_FILE							0x0404
#define ROM_KERNEL_CMD_ERROR_STATUS				0x0505
#define RAM_KERNEL_CMD_HEADER							0x0606
//#define ROM_KERNEL_CMD_RE_ENUM 0x0909
#define ROM_KERNEL_CMD_DCD_WRITE					0x0A0A
#define ROM_KERNEL_CMD_JUMP_ADDR					0x0B0B
#define ROM_KERNEL_CMD_SKIP_DCD_HEADER				0x0C0C

nxpfrankli, 2020-04-02:
I am checking with ROM team. Supposed it should be done automatically.

ldts, 2020-04-02:
thanks @nxpfrankli !! I'll try to add some more details below in case it helps

System description on imx7ulp
==============================
We are trying to boot an imx7ulp _closed_ system; the configuration we
use requires that the CAAM remains unlocked during boot so the trust
zone can reconfigure it.

We followed the recommendations from the Code Signing Tool 3.3 and
added the following command to the CSF template used during the
signing of the image

[Unlock]
Engine = CAAM
Features = MID, RNG

Using the hab_csf_parser on the generated image we indeed find the
HAB command in the CSF region:

CSF TAG: 0xB2 [Unlock]
  Parsing following values
  0xB2 0x00 0x08 0x1D 0x00 0x00 0x00 0x03

decodes as:

TAG     VALUE           DEFINITION              DESCRIPTION
----------------------------------------------------------------------
        0xB2            HAB_CMD_UNLK            Unlock Command

LEN     VALUE           DEFINITION              DESCRIPTION
----------------------------------------------------------------------
        0x8             8 bytes                 Length of Unlock Command

ENG     VALUE           DEFINITION              DESCRIPTION
----------------------------------------------------------------------
        0x1D            HAB_ENG_CAAM            CAAM

FEATURE VALUE           DEFINITION              DESCRIPTION
----------------------------------------------------------------------
        0x3             BITWISE OR              Bitwise OR of features below

        0x1             UNLOCK_CAAM_MID         Unlock CAAM MID feature enabled

        0x2             UNLOCK_CAAM_RNG         Unlock CAAM RNG feature enabled

When the system boots, SPL kicks off the Trust Zone (OP-TEE) and
OP-TEE configures the CAAM. Part of the CAAM configuration is the
initialization of the job rings: and the first step in the job ring
initialization is to set the job ring ownership.

However, to set the job ring ownership, we first check if the MIDs
have been locked by the ROM (bits 31 of JR0MID_MS@0x10 and
JR1MID_MS@18) on the CAAM (@0x8000000).

Problem statement
=================

1) When the system boots from eMMC, the MID is effectively unlocked so
the HAB is working as it should.

2) However when the system boots via SDP, the MID remains locked and we
therefore fail to initialize the CAAM in the Trust Zone.

SDP boot
========
Booting from SDP requires that we write the whole SPL image (having
cleared the DCD address), then write the original IVT header (size
0x400) and then jump to the SELF address.

IVT : HEADER    = 0x402000D1
      START     = 0x2F011000
      RES1      = 0x00000000
      DCD       = 0x2F01042C
      BOOT DATA = 0x2F010420
      SELF      = 0x2F010400
      CSF       = 0x2F032000 <-- contains the HAB commands
      RES2      = 0x00000000

In terms of SDP commands, the mfgtool will issue:

- ROM_KERNEL_CMD_WR_FILE (SPL image)     -- 0x0404
- ROM_KERNEL_CMD_WR_FILE (IVT header)    -- 0x0404
- ROM_KERNEL_CMD_JUMP_ADDR @ 0x2F010400  -- 0x0B0B

Somehow this sequence of calls do boot the SPL on a closed device (so
the keys in the CSF must be being validated) by the HAB but the CAAM
MID remains locked.

Questions
=========
does the SDP protocol need some extra command in order to force the
unlocking of the CAAM? if not, could we be hitting some timing issue
in the HAB?

nxpfrankli, 2020-04-02:
I remember 7D don't need cleared the DCD address, just send skip_dcd command before jump.

nxpfrankli, 2020-04-03:
ROM team think maybe your image have problem, are you use exact the same one wIth eMMC.
what your uuu command to download spl?

ldts, 2020-04-03:
@nxpfrankli yes I use the exact same binary (but sign it differently for MMC boot versus SDP boot)

I attach the csf_parser analisis on both the images. 

Booting from SDP fails to unlock the CAAM where as booting from MMC unlocks it successfully.

Booting from SDP:
boot -f SPL.sdp -dcdaddr 0x2f010000 -cleardcd

[signed.for.mmc.zip](https://github.com/NXPmicro/mfgtools/files/4426082/signed.for.mmc.zip)
[signed.for.sdp.zip](https://github.com/NXPmicro/mfgtools/files/4426083/signed.for.sdp.zip)

ldts, 2020-04-03:
Also notice that for MMC the csf config template will look as 

> [Header]
> Version = 4.1
> Hash Algorithm = sha256
> Engine Configuration = 0
> Certificate Format = X509
> Signature Format = CMS
> Engine = CAAM
> 
> [Install SRK]
> File = "./SRK_1_2_3_4_table.bin"
> Source index = 0
> 
> [Install CSFK]
> File = "./CSF_1_crt.pem"
> 
> [Authenticate CSF]
> 
> [Unlock]
> Engine = CAAM
> Features = MID, RNG
> 
> [Install Key]
> Verification index = 0
> Target index = 2
> File = "./IMG_1_crt.pem"
> [Authenticate Data]
> Verification index = 2
> Blocks = 0x2f010400 0x00000000 0x00021c00 "SPL.mod"

where as for SDP

> [Header]
> Version = 4.1
> Hash Algorithm = sha256
> Engine Configuration = 0
> Certificate Format = X509
> Signature Format = CMS
> Engine = CAAM
> 
> [Install SRK]
> File = "./SRK_1_2_3_4_table.bin"
> Source index = 0
> 
> [Install CSFK]
> File = "./CSF_1_crt.pem"
> 
> [Authenticate CSF]
> 
> [Unlock]
> Engine = CAAM
> Features = MID, RNG
> 
> [Install Key]
> Verification index = 0
> Target index = 2
> File = "./IMG_1_crt.pem"
> [Authenticate Data]
> Verification index = 2
> Blocks = 0x2f010000 0x0000002c 0x00000258 "SPL.mod"
> 
> [Authenticate Data]
> Verification index = 2
> Blocks = 0x2f010400 0x00000000 0x00021c00 "SPL.mod"

nxpfrankli, 2020-04-03:
why sign is difference?

ldts, 2020-04-03:
Because once the device has been closed, SDP requires that the CSF is modifed to include a check for the DCD table. SDP also requires that the DCD address is cleared from the header 

we added some documentation here:
https://docs.foundries.io/latest/reference/secure-boot.html#how-to-sign-an-spl-image-for-sdp-ii

which matches pretty much what other have documented
https://boundarydevices.com/high-assurance-boot-hab-dummies/

Notice that **we do boot over SDP** (so signing must be correct or we wouldnt boot): the only issue we face is that HAB remains closed despite the request for unlock....

nxpfrankli, 2020-04-03:
Old chip require clear DCD address. but new chip needn't, just need send SKIP_DCD_CMD before JUMP cmd.  
ROM team said no difference.  Can you try exact one with SKIP_DCD_CMD before jump?

ldts, 2020-04-03:
ah cool ; so to be clear you are saying that I do as follows:

1) sign the code for SDP in the same way than we do for MMC
2) enable ROM_INFO_HID_SKIP_DCD 
3) boot without the cleardcd flag.

did I get it right?

nxpfrankli, 2020-04-03:
Yes

nxpfrankli, 2020-04-03:
Do  you use yourself tools? "uuu sdp: boot" already consider this

ldts, 2020-04-03:
Ok so run these tests and I can see the system boot - so you are right and we dont need to clear the DCD address... but still no unlock of the CAAM

> uuu_version 1.0.1                                                                                                                                                                                                                                                                                                                                                                       
>                                                                                                                                                                                                                                                                                                                                                                                         
> SDP: boot -f SPL-imx7ulpea-ucom.mmc                                                                                                                                                                                                                                                                                                                                               
>                                                                                                                                                                                                                                                                                                                                                                                         
> SDPU: delay 1000                                                                                                                                                                                                                                                                                                                                                                        
> SDPU: write -f u-boot-imx7ulpea-ucom.itb                                                                                                                                                                                                                                                                                                                                          
> SDPU: jump                                                                                                                                                                                                                                                                                                                                                                              
>                                                                                                                                                                                                                                                                                                                                                                                         
> FB: done  

the output remains the same:

> U-Boot SPL 2019.10 (Apr 02 2020 - 17:53:37 +0000)                                                                                                                                                                                                                                                                                                                                       
> Trying to boot from USB SDP                                                                                                                                                                                                                                                                                                                                                             
> usb dr_mode not found                                                                                                                                                                                                                                                                                                                                                                   
> USB EHCI 1.00                                                                                                                                                                                                                                                                                                                                                                           
> SDP: initialize...                                                                                                                                                                                                                                                                                                                                                                      
> SDP: handle requests...                                                                                                                                                                                                                                                                                                                                                                 
[... cut ...]                                                                                                   
> E/TC:0 0 Call stack:                                                                                                                                                                                                                                                                                                                                                                    
> E/TC:0 0  0x9e00532d

ldts, 2020-04-03:
Booting from flash:

> U-Boot SPL 2019.10 (Apr 02 2020 - 17:53:37 +0000)                                                                                                                                                                                                                                                                                                                                       
> Trying to boot from MMC1                                                                                                                                                                                                                                                                                                                                                                
>  Checking hash(es) for Image optee ... sha256+ OK                                                                                                                                                                                                                                                                                                                                     
>  Checking hash(es) for Image uboot ... sha256+ OK                                                                                                                                                                                                                                                                                                                                     
>  Checking hash(es) for Image ubootfdt ... sha256+ OK                                                                                                                                                                                                                                                                                                                                  
> I/TC:                                                                                                                                                                                                                                                                                                                                                                                   
> I/TC: Non-secure external DT found                                                                                                                                                                                                                                                                                                                                                      
> I/TC: imx_lpuart: device parameters ignored (115200n8)                                                                                                                                                                                                                                                                                                                                  
> I/TC: Switching console to device: /ahb-bridge0@40000000/serial@402D0000                                                                                                                                                                                                                                                                                                                
> I/TC: OP-TEE version: 3.6.0-177-gcef93f4d-dev (gcc version 9.2.0 (GCC)) #16 Fri Apr  3 08:49:30 UTC 2020 arm                                                                                                                                                                                                                                                                            
> E/TC:0 0 caam_hal_jr_setowner:34 JR0MIDR_MS value 0x8001                                                                                                                                                                                                                                                                                                                                
> E/TC:0 0 caam_hal_jr_setowner:34 JR1MIDR_MS value 0x8001                                                                                                                                                                                                                                                                                                                                
> I/TC: Initialized                                                                                                                                                                                                                                                                                                                                                                       
> I/TC: Non-secure external DT found                                                                                                                                                                                                                                                                                                                                                      
>                                          

booting from SDP

[... cut ...]                                                                                     
> I/TC: Initialized                                                                                                                                                                                                                                                                                                                                                                       
> I/TC: Non-secure external DT found

ldts, 2020-04-03:
ah fixed!!!
just removed the delay from the boot script....

nxpfrankli, 2020-04-03:
Remove delay fix problem?

ldts, 2020-04-03:
sorry no....I was booting from mmc.

ldts, 2020-04-04:
I also tried to use the hab_log_parser but nothing gets reported on the data retrieved from u-boot.

jramirez@trex bin (master %)$ ./hab_log_parser -a ./uboot_persistent_ascii.txt
Output selected as Standard Output

[image]

is there anything I can do from my end? Any command I can invoke in CSF that is perhaps not documented to force the HAB to reinit the CAAM and unlock it?

ldts, 2020-04-06:
@nxpfrankli maybe it is a bug in HAB?

nxpfrankli, 2020-04-06:
HAB don't know if boot from USB or EMMC.

ldts, 2020-04-06:
then SDP on the ROM side? 
We are transferring the signed image with the unlock request and receiving an OK on the transfer.

why would  HAB not unlock or raise some error event on the request?

ldts, 2020-04-20:
We were able to test on an open board, and the CAAM did unlock successfully when using SDP. Is is possible that we are missing something else in the CSF? something specific to closed boards?
@nxpfrankli ^^?

nxpfrankli, 2020-04-21:
let's ping ROM team again.

ldts, 2020-04-30:
@nxpfrankli any feedback?

nxpfrankli, 2020-04-30:
Sorry, not get feedback yet.  May1-5 is china holiday. I can ping after May 6

ldts, 2020-05-01:
sure, thanks for following up!

nxpfrankli, 2020-05-12:
After investigated and It may be ROM bug.  Do you know chip version do you used?

ldts, 2020-05-12:
thanks for following up @nxpfrankli : the SoC tag I can read on top of the chip is PC.IMX7U.5DVP08SC

nxpfrankli, 2020-05-12:
Generally, uboot print

U-Boot 2020.04-5.4.24-2.1.0+g0cd21c2391 (Apr 28 2020 - 22:02:17 +0000)

CPU:   i.MX8MNano Quad **rev1.0** 1500 MHz (running at 1200 MHz)

Can you tell us revXXX?

mike-scott, 2020-05-12:
(co-worker of @ldts )
We see: `CPU:   Freescale i.MX7ULP rev2.1 at 500 MHz`

mike-scott, 2020-05-12:
@nxpfrankli ^^

ldts, 2020-06-17:
@nxpfrankli were you able to confirm this ?

nxpfrankli, 2020-06-17:
Yes, it is ROM issue.

ldts, 2020-06-17:
ok. thanks @nxpfrankli for following up on this matter!

closed 2020-06-17 by ldts