nxp-imx/mfgtools #183: UUU libusb: error ubuntu 20.04

Hadatko, 2020-05-04:
Hello, please add support for new ubuntu LTS. After upgrading to ubuntu 20.04 LTS uuu stopped working (on both machine- both were upgraded). Throwing libusb: error. With verbose -V cmd argument it works sometimes (50/50).

Hadatko, 2020-05-04:
Fixed with using "-pp 1"

qcabrol, 2020-05-26:
Actually I could not flash either with my Ubuntu 20.04 with the -pp 1 option either. Various option didn't change a thing. Will be posting the verbose results here today.

nxpfrankli, 2020-05-27:
It work well in my ubuntu 20.04. Not clue yet

qcabrol, 2020-06-11:
```
 develop ●  sudo mfgtools-signed/uuu -pp 1  -V mfgtools-signed/m4app.uuu

uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.102-9-gfdf71fa

Build in config:
        Pctl     Chip            Vid     Pid     BcdVersion
        ==================================================
        SDPS:    MX8QXP          0x1fc9  0x012f  [0x0002..0xffff]
        SDPS:    MX8QM           0x1fc9  0x0129  [0x0002..0xffff]
        SDPS:    MX28            0x15a2  0x004f
        SDPS:    MX815           0x1fc9  0x013e
        SDP:     MX7D            0x15a2  0x0076
        SDP:     MX6Q            0x15a2  0x0054
        SDP:     MX6D            0x15a2  0x0061
        SDP:     MX6SL           0x15a2  0x0063
        SDP:     MX6SX           0x15a2  0x0071
        SDP:     MX6UL           0x15a2  0x007d
        SDP:     MX6ULL          0x15a2  0x0080
        SDP:     MX6SLL          0x1fc9  0x0128
        SDP:     MX7ULP          0x1fc9  0x0126
        SDP:     MXRT106X        0x1fc9  0x0135
        SDP:     MX8MM           0x1fc9  0x0134
        SDP:     MX8MQ           0x1fc9  0x012b
        SDPU:    SPL             0x0525  0xb4a4  [0x0000..0x04ff]
        SDPV:    SPL1            0x0525  0xb4a4  [0x0500..0x9998]
        SDPU:    SPL             0x0525  0xb4a4  [0x9999..0x9999]
        FBK:                     0x066f  0x9afe
        FBK:                     0x066f  0x9bff
        FB:                      0x0525  0xa4a5
        FB:                      0x18d1  0x0d02
Wait for Known USB Device Appear...
New USB Device Attached at 1:11
1:11>Start Cmd:SDP: boot -f SPL-aeler-imx7ulpea-ucom
1:11>Fail HID(W):LIBUSB_ERROR_TIMEOUT(1.001s)
```

qcabrol, 2020-06-11:
I tried to edit the udev rules as suggested in using uuu -udev
mfgtools-signed/uuu -v -pp 1  mfgtools-signed/m4app.uuu      
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.102-9-gfdf71fa

```
Build in config:
        Pctl     Chip            Vid     Pid     BcdVersion
        ==================================================
        SDPS:    MX8QXP          0x1fc9  0x012f  [0x0002..0xffff]
        SDPS:    MX8QM           0x1fc9  0x0129  [0x0002..0xffff]
        SDPS:    MX28            0x15a2  0x004f
        SDPS:    MX815           0x1fc9  0x013e
        SDP:     MX7D            0x15a2  0x0076
        SDP:     MX6Q            0x15a2  0x0054
        SDP:     MX6D            0x15a2  0x0061
        SDP:     MX6SL           0x15a2  0x0063
        SDP:     MX6SX           0x15a2  0x0071
        SDP:     MX6UL           0x15a2  0x007d
        SDP:     MX6ULL          0x15a2  0x0080
        SDP:     MX6SLL          0x1fc9  0x0128
        SDP:     MX7ULP          0x1fc9  0x0126
        SDP:     MXRT106X        0x1fc9  0x0135
        SDP:     MX8MM           0x1fc9  0x0134
        SDP:     MX8MQ           0x1fc9  0x012b
        SDPU:    SPL             0x0525  0xb4a4  [0x0000..0x04ff]
        SDPV:    SPL1            0x0525  0xb4a4  [0x0500..0x9998]
        SDPU:    SPL             0x0525  0xb4a4  [0x9999..0x9999]
        FBK:                     0x066f  0x9afe
        FBK:                     0x066f  0x9bff
        FB:                      0x0525  0xa4a5
        FB:                      0x18d1  0x0d02
Wait for Known USB Device Appear...
New USB Device Attached at 1:11
1:11>Start Cmd:SDP: boot -f SPL-aeler-imx7ulpea-ucom
6400%1:11>Okay (2.05s)
```

I see that log and it hangs here, nothing is being flashed to the device and it never exits.

qcabrol, 2020-06-11:
```
 6379.447818] usb 1-1.1: new high-speed USB device number 16 using xhci_hcd
[ 6379.468849] usb 1-1.1: New USB device found, idVendor=1fc9, idProduct=0126, bcdDevice= 0.01
[ 6379.468854] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 6379.468858] usb 1-1.1: Product: SP Blank ULP1
[ 6379.468860] usb 1-1.1: Manufacturer:    NXP    SemiConductor Inc 
[ 6379.472398] hid-generic 0003:1FC9:0126.000B: hiddev0,hidraw1: USB HID v1.10 Device [   NXP    SemiConductor Inc  SP Blank ULP1] on usb-0000:00:14.0-1.1/input0
[ 6529.738439] usb 1-1.1: USB disconnect, device number 16
```

nxpfrankli, 2020-06-11:
look like SPL-aeler-imx7ulpea-ucom is not boot.  USB still in ROM PID\VID.  SPL should use difference PID\VID

qcabrol, 2020-06-12:
Correct:

qcabrol, 2020-06-30:
The Same SPL are used under Ubuntu18.04 successfully though with the same uuu calls.

qcabrol, 2020-07-20:
This seems operational downloading uuu latest release !

qcabrol, 2020-07-20:
Result with latest uuu release:

```
Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
Wait for Known USB Device Appear...
New USB Device Attached at 1:12
1:12>Start Cmd:SDP: boot -f SPL-aeler-imx7ulpea-ucom
22%libusb: error [udev_hotplug_event] ignoring udev action bind
6400%1:12>Okay (0.265s)
New USB Device Attached at 1:12
1:12>Start Cmd:SDPU: delay 1000
libusb: error [udev_hotplug_event] ignoring udev action bind
1:12>Okay (1s)
1:12>Start Cmd:SDPU: write -f u-boot-aeler-imx7ulpea-ucom.itb
99%1:12>Okay (0.346s)
1:12>Start Cmd:SDPU: jump
6400%1:12>Okay (1.001s)
New USB Device Attached at 1:12
1:12>Start Cmd:FB: ucmd setenv fastboot_buffer ${initrd_addr}
1:12>Okay (0.005s)
1:12>Start Cmd:FB: download -f ../sdk20-app_flash.img
libusb: error [udev_hotplug_event] ignoring udev action bind
1:12>Okay (0.012s)
1:12>Start Cmd:FB: ucmd sf probe
1:12>Okay (0.013s)
1:12>Start Cmd:FB[-t 40000]: ucmd sf erase 0 40000
1:12>Okay (2.145s)
1:12>Start Cmd:FB: ucmd sf write ${fastboot_buffer} 0 ${fastboot_bytes}
1:12>Okay (1.014s)
1:12>Start Cmd:FB: done
1:12>Okay (0s)

```

qcabrol, 2020-07-20:
We observed that in fact the uuu tool works as epxected on 20.04 as long as the device is not in a closed state (burned fused) After that, subsequent attempts at uuu calls to flash M4 or A7 program fails with a HW libust tiemout issue.

nxpfrankli, 2020-07-20:
Did your closed part stop here?

1:12>Start Cmd:SDP: boot -f SPL-aeler-imx7ulpea-ucom
22%libusb: error [udev_hotplug_event] ignoring udev action bind
6400%1:12>Okay (0.265s)

look like autherization failure. Are you use signed image to download?

qcabrol, 2020-07-20:
Apologies, the capture above was from a flashing of the M4 program before the device was closed.
And from the device trace after booting it at that point, this went ok.

After the device was closed though, here is what I got:

```
 mfgtools-signed/uuu -pp 1 mfgtools-signed/full_tar.uuu
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.191-0-g4fe24b9
Success 0    Failure 1                                                                                                                                                                                            
1:12     1/ 1 [HID(W):LIBUSB_ERROR_TIMEOUT           ] SDP: boot -f SPL-aeler-imx7ulpea-ucom   
```

Note that reverting to Ubuntu 18.04 with the exact same files (uboot.itb, signed SPLs, signed Linux image and M4 program), everything goes as expected. So we are sticking to 18.04 for that reason at the moment.

nxpfrankli, 2020-07-20:
I don't know what happen now. it is very strange

qcabrol, 2020-07-27:
Trying on a different 20.04 machine the tool worked well.
So it might be an issue at USB driver level.
Detail of the machine on which the issue was observed: ASUS Zenbook UX550-GE.

Hence, diving deeper, it seems it is an Ubuntu issue not an issue directly linked to uuu.
Thanks for your support !

rob-mclean, 2020-08-28:
I'm having problems with UUU also.  Similar but different.  Instead of the  "LIBUSB_ERROR_TIMEOUT", I'm getting a "LIBUSB_ERROR_NO_DEVICE".  
I pulled the latest version of the uuu code and after some mods, I was able to see HIDTrans::write successfully transferred data for the first 8 packets of my u-boot image.  On the ninth packet I got the "NO_DEVICE" error, which seems to indicate something happened to cause the device to be disconnected after the 8th packet.
My host computer also has an ASUS motherboard, so I'm suspicious that might be the commonality.
I'm going to try downgrading to Ubuntu 18.04LTS and see if that fixes my problems as well.

closed 2021-03-10 by nxpfrankli

qcabrol, 2022-03-31:
Hello @nxpfrankli 

We have been using older hardware (old Thinkpad and Alienware M17x from 2012) to flash closed devices ever since the problem was experienced but it is not a viable solution long term. We have tried again with more recent versions of the uuu tool and the issur persist.

We are running on closed i.Mx7-ULP boards.
Recent tests have been conducted on Ubuntu 20.04 ASUS Zenbook UX550 and on an Dell XPS 15 from last year.
Problem is present on both machines regardless of the use of USB Hubs.

Under windows our manufacturers have been facing the same issues for our first series production but it seems the use of a USB hub helped (it degrades the USB driver that is used to USB 2.0 if I am correct).

could we re-open this ticket ? We depend on this tool proper operation for series production.
And the outcome should be independent of the hardware running the uuu.

[image]

[image]

qcabrol, 2022-03-31:
Example the script works on:
`Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   39 bits physical, 48 bits virtual
CPU(s):                          8
On-line CPU(s) list:             0-7
Thread(s) per core:              2
Core(s) per socket:              4
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           60
Model name:                      Intel(R) Core(TM) i7-4800MQ CPU @ 2.70GHz
Stepping:                        3
CPU MHz:                         2852.015
CPU max MHz:                     3700,0000
CPU min MHz:                     800,0000
BogoMIPS:                        5387.58
Virtualization:                  VT-x
L1d cache:                       128 KiB
L1i cache:                       128 KiB
L2 cache:                        1 MiB
L3 cache:                        6 MiB
NUMA node0 CPU(s):               0-7
Vulnerability Itlb multihit:     KVM: Mitigation: VMX disabled
Vulnerability L1tf:              Mitigation; PTE Inversion; VMX conditional cache flushes, SMT vulnerable
Vulnerability Mds:               Mitigation; Clear CPU buffers; SMT vulnerable
Vulnerability Meltdown:          Mitigation; PTI
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; Retpolines, IBPB conditional, IBRS_FW, STIBP conditional, RSB filling
Vulnerability Srbds:             Mitigation; Microcode
Vulnerability Tsx async abort:   Not affected
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm ida arat pln pts md_clear flush_l1d`

Does not work on:

`Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   39 bits physical, 48 bits virtual
CPU(s):                          12
On-line CPU(s) list:             0-11
Thread(s) per core:              2
Core(s) per socket:              6
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       GenuineIntel
CPU family:                      6
Model:                           158
Model name:                      Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz
Stepping:                        10
CPU MHz:                         800.155
CPU max MHz:                     4100.0000
CPU min MHz:                     800.0000
BogoMIPS:                        4399.99
Virtualization:                  VT-x
L1d cache:                       192 KiB
L1i cache:                       192 KiB
L2 cache:                        1.5 MiB
L3 cache:                        9 MiB
NUMA node0 CPU(s):               0-11
Vulnerability Itlb multihit:     KVM: Mitigation: Split huge pages
Vulnerability L1tf:              Mitigation; PTE Inversion; VMX conditional cache flushes, SMT vulnerable
Vulnerability Mds:               Mitigation; Clear CPU buffers; SMT vulnerable
Vulnerability Meltdown:          Mitigation; PTI
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:        Mitigation; Full generic retpoline, IBPB conditional, IBRS_FW, STIBP conditional, RSB filling
Vulnerability Srbds:             Mitigation; Microcode
Vulnerability Tsx async abort:   Not affected
Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon
                                  pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcn
                                 t tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgs
                                 base tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_
                                 epp md_clear flush_l1d
`

nxpfrankli, 2022-03-31:
It is quite difficult to debug the problem. 

It may related USB host controller driver.

Can you try run 
uuu SDP: dcd -f uboot

see what happen at your failure machine.

qcabrol, 2022-03-31:
[image]

nxpfrankli, 2022-03-31:
Can you sperate boot cmd to 3 commands?

SDP: dcd -f uboot
SDP: write -f uboot -ivt 0
SDP: jump -f uboot -ivt 0

qcabrol, 2022-03-31:
Might be missing something here but it tells me "no valid IVT header found"

nxpfrankli, 2022-03-31:
using spl_mfgtool.signed instead of u-boot-mfgtool.itb

qcabrol, 2022-04-07:
Test with spl_mfgtool.signed

```
sudo ./uuu -v SDP: dcd -f SPL-mfgtool.signed
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.165-0-ge10b026

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
Wait for Known USB Device Appear...
>Start Cmd:SDP: dcd -f SPL-mfgtool.signed
Wait for Known USB |New USB Device Attached at 1:1
6400%1:1>Fail HID(W):LIBUSB_ERROR_TIMEOUT(5.214s)

Error: HID(W):LIBUSB_ERROR_TIMEOUT
```

```
sudo ./uuu -v SDP: write -f SPL-mfgtool.signed -ivt 0
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.165-0-ge10b026

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
Wait for Known USB Device Appear...
>Start Cmd:SDP: write -f SPL-mfgtool.signed -ivt 0
Wait for Known USB /New USB Device Attached at 1:1
1%1:1>Fail HID(W):LIBUSB_ERROR_TIMEOUT(4.609s)

Error: HID(W):LIBUSB_ERROR_TIMEOUT
```

```
sudo ./uuu -v SDP: jump -f SPL-mfgtool.signed -ivt 0
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.165-0-ge10b026

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
Wait for Known USB Device Appear...
>Start Cmd:SDP: jump -f SPL-mfgtool.signed -ivt 0
Wait for Known USB \New USB Device Attached at 1:1
6400%1:1>Okay (3.213s)

Okay
```

qcabrol, 2022-04-07:
With -pp 1 option

```
sudo ./uuu -v -pp 1 SDP: dcd -f SPL-mfgtool.signed
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.165-0-ge10b026

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
Wait for Known USB Device Appear...
>Start Cmd:SDP: dcd -f SPL-mfgtool.signed
Wait for Known USB -New USB Device Attached at 1:2
168%1:2>Okay (2.21s)

Okay
```
```
sudo ./uuu -v -pp 1  SDP: write -f SPL-mfgtool.signed -ivt 0
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.165-0-ge10b026

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
Wait for Known USB Device Appear...
>Start Cmd:SDP: write -f SPL-mfgtool.signed -ivt 0
Wait for Known USB /New USB Device Attached at 1:2
1%1:2>Fail HID(W):LIBUSB_ERROR_TIMEOUT(3.808s)

Error: HID(W):LIBUSB_ERROR_TIMEOUT
```
```
sudo ./uuu -v -pp 1 SDP: jump -f SPL-mfgtool.signed -ivt 0
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.165-0-ge10b026

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
Wait for Known USB Device Appear...
>Start Cmd:SDP: jump -f SPL-mfgtool.signed -ivt 0
Wait for Known USB \New USB Device Attached at 1:2
6400%1:2>Okay (2.41s)

Okay
```

So the *write* command seems to be the problematic one

Hadatko, 2022-04-09:
I know i am not able to use this tool with usb hub.

nxpfrankli, 2022-04-10:
So far only you report hub problem, our internal test have not problem with hub.

qcabrol, 2022-05-11:
@nxpfrankli The issue persists with and without a hub on my side for most ubuntu computers.
See April 7th traces you asked for.

nxpfrankli, 2022-05-11:
The current clue can't help me identify the problem.