nxp-imx/mfgtools #185: uuu-Determine Device From Script

dornenvogel, 2020-05-05:
Hi,
we are using uuu tool for several i.MX based boards, e.g. i.MX6 Solo and i.MX6 ULL. The former mfg tool had a nice feature to ask for the device from within the script:

<CMD state="BootStrap" type="jump" ifdev="MX6D"> ERROR : Wrong software. Do not program i.MX6 Solo software to i.MX6 ULL !!! </CMD>

So this line prevented us from using the wrong script for a certain device. This is especially important as we are burning fuses from that script and the OTP fuses are different in our projects.

Question: Is there a way in uuu of determining the device, e.g. by VID/PID from within our scripts so that script is exiting when the data is wrong?

Thank you!

nxpfrankli, 2020-05-05:
Not yet.

nxpfrankli, 2021-06-28:
I add very simple logic check. Please try latest one

`sdp: if @chip@ != mx7dx then error ` follow error message . Error is command, return error to user

closed 2021-07-28 by nxpfrankli