nxp-imx/mfgtools #186: LIBUSB_ERROR_TIMEOUT on Ubuntu 18.04.4 LTS, uuu 1.3.71 and i.MX8MM EVK

rogeriorps, 2020-05-05:
Hi, I'm trying to burn u-boot on i.MX8MM EVK emmc using Ubuntu 18.04.4 LTS on host machine. It's returning the following error:

Wait for Known USB Device Appear...
New USB Device Attached at 1:2
1:2>Start Cmd:SDP: boot -f u-boot.imx
**1:2>Fail HID(W):LIBUSB_ERROR_TIMEOUT(1s)**

Do you have any clue on what's wrong or any advice?

Here is the complete log:

rogerio@rogerio-Latitude-7490:~/uuu$ sudo ./uuu_1_3_71 -V -b emmc u-boot.imx 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.171-0-g4249f93

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001

Run built-in script:
 
uuu_version 1.2.39

# @u-boot.imx            | bootloader
# @u-boot.imx   [u-boot.imx] | image burn to emmc, default is the same as bootloader

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f u-boot.imx

# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM
SDPS: boot -f u-boot.imx

# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
# {
SDPU: delay 1000
SDPU: write -f u-boot.imx -offset 0x57c00
SDPU: jump
# }

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f u-boot.imx -skipspl
SDPV: jump
# }

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${emmc_dev}
FB: ucmd mmc dev ${emmc_dev}
FB: flash bootloader u-boot.imx
FB: ucmd if env exists emmc_ack; then ; else setenv emmc_ack 0; fi;
FB: ucmd mmc partconf ${emmc_dev} ${emmc_ack} 1 0
FB: Done

Wait for Known USB Device Appear...
New USB Device Attached at 1:2
1:2>Start Cmd:SDP: boot -f u-boot.imx
1:2>Fail HID(W):LIBUSB_ERROR_TIMEOUT(1s)

Best regards,
Rogerio

nxpfrankli, 2020-05-05:
8MM need use SPL. it can't direct download uboot. 
You use imx_mkimage to package, SPL, DDR firmware and uboot.

rogeriorps, 2020-05-05:
Hi Frank, yes, it's working now. I thought it could be related to my Ubuntu libusb version.
Thanks for your help!
Rogerio

closed 2020-05-05 by rogeriorps