nxp-imx/mfgtools #188: example_kernel_emmc.uuu not work for L4.14.98_2.3.0_images_MX8QXPMEK

pingwuu, 2020-05-13:
During developing stage, the kernel and dtb is offen modified and test repeat. so I try to using example_kernel_emmc.uuu to flash bootloader,kernel and rootfs. but after example_kernel_emmc.uuu done, then u-boot can not find dtb and kernel well.
Reproduce step:
1. download L4.14.98_2.3.0_images_MX8QXPMEK.zip  (https://www.nxp.com/webapp/Download?colCode=L4.14.98_2.3.0_MX8QXP&appType=license)
2. modify example_kernel_emmc.uuu to disable optee image  (details example_kernel_emmc.uuu file attached  [example_kernel_emmc.zip](https://github.com/NXPmicro/mfgtools/files/4621953/example_kernel_emmc.zip)
)
3. set imx8qxp mek board(B0) boot mode SW2 0001 as serial download
4. execute uuu.exe example_kernel_emmc.uuu
5. set boot mode SW2 0010 as eMMC boot
6. reboot board
7. u-boot can boot, but can not boot kernel correctly.

Any mistake or miss some steps?  could you help give some tip for this error? thanks.

///////////////////////////////////reboot log/////////////////////////////////////////////////

U-Boot 2018.03-4.14.98-2.3.0+g0e207921e9 (Dec 19 2019 - 04:18:59 +0000)

CPU:   Freescale i.MX8QXP revB A35 at 1200 MHz at 34C
Model: Freescale i.MX8QXP MEK
Board: iMX8QXP MEK
Boot:  MMC0
DRAM:  3 GiB
TCPC:  Vendor ID [0x1fc9], Product ID [0x5110], Addr [I2C1 0x50]
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
Loading Environment from MMC... *** Warning - bad CRC, using default environment

Failed (-5)
[pcie_ctrlb_sata_phy_init_rc] LNK DOWN 8600000
In:    serial
Out:   serial
Err:   serial

 BuildInfo:
  - SCFW 5a35e406, SECO-FW f449a0d2, IMX-MKIMAGE 25560004, ATF 09c5cc9
  - U-Boot 2018.03-4.14.98-2.3.0+g0e207921e9

switch to partitions #0, OK
mmc0(part 0) is current device
flash target is MMC:0
Net:
Warning: ethernet@5b040000 using MAC address from ROM
eth0: ethernet@5b040000 [PRIME]
Warning: ethernet@5b050000 (eth1) using random MAC address - 1a:3f:b5:9d:d5:27
, eth1: ethernet@5b050000
Fastboot: Normal
Normal Boot
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0(part 0) is current device
** Unable to read file boot.scr **
** Unable to read file Image **
Booting from net ...
ethernet@5b040000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
Could not initialize PHY ethernet@5b040000
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 7
BOOTP broadcast 8
BOOTP broadcast 9
BOOTP broadcast 10
BOOTP broadcast 11
BOOTP broadcast 12
BOOTP broadcast 13
BOOTP broadcast 14
BOOTP broadcast 15
BOOTP broadcast 16
BOOTP broadcast 17

Retry time exceeded; starting again
ethernet@5b040000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
Could not initialize PHY ethernet@5b040000
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 7
BOOTP broadcast 8
BOOTP broadcast 9
BOOTP broadcast 10
BOOTP broadcast 11
BOOTP broadcast 12
BOOTP broadcast 13
BOOTP broadcast 14
BOOTP broadcast 15
BOOTP broadcast 16
BOOTP broadcast 17

Retry time exceeded; starting again
WARN: Cannot load the DT
=>

nxpfrankli, 2020-05-13:
can you run at uboot console.
fatls mmc 0:1

pingwuu, 2020-05-14:
@nxpfrankli ,
pls check info as below, thanks.

================
=> fatls mmc 0:1
 23376384   Image-imx8_all.bin
    83922   Image-fsl-imx8qxp-mek.dtb

2 file(s), 0 dir(s)

=>
=======================
and I attach printenv info as below:
==========
=> printenv
auth_os=auth_cntr ${cntr_addr}
baudrate=115200
board_name=MEK
board_rev=iMX8QXP
boot_fdt=try
boot_os=booti ${loadaddr} - ${fdt_addr};
bootargs=console=ttyLP0,115200 earlycon root=/dev/nfs ip=dhcp nfsroot=:,v3,tcp
bootcmd=mmc dev ${mmcdev}; if mmc rescan; then if run loadbootscript; then run boot                                                                                                          script; else if test ${sec_boot} = yes; then if run loadcntr; then run mmcboot; els                                                                                                          e run netboot; fi; else if run loadimage; then run mmcboot; else run netboot; fi; f                                                                                                          i; fi; else booti ${loadaddr} - ${fdt_addr}; fi
bootcmd_mfg=run mfgtool_args;if iminfo ${initrd_addr}; then if test ${tee} = yes; t                                                                                                          hen bootm ${tee_addr} ${initrd_addr} ${fdt_addr}; else booti ${loadaddr} ${initrd_a                                                                                                          ddr} ${fdt_addr}; fi; else echo "Run fastboot ..."; fastboot 0; fi;
bootdelay=3
bootscript=echo Running bootscript from mmc ...; source
cntr_addr=0x98000000
cntr_file=os_cntr_signed.bin
commit_atf=09c5cc9
commit_mkimage=25560004
commit_scfw=5a35e406
commit_secofw=f449a0d2
console=ttyLP0
dom0fdt_file=fsl-imx8qxp-mek-dom0.dtb
emmc_dev=0
ethact=ethernet@5b040000
ethaddr=00:04:9f:05:c8:da
ethprime=eth0
fastboot_dev=mmc0
fdt_addr=0x83000000
fdt_file=fsl-imx8qxp-mek.dtb
fdt_high=0xffffffffffffffff
fdtcontroladdr=fd667208
get_cmd=dhcp
image=Image
initrd_addr=0x83100000
initrd_high=0xffffffffffffffff
jh_mmcboot=setenv fdt_file fsl-imx8qxp-mek-root.dtb;setenv boot_os 'scu_rm dtb ${fd                                                                                                          t_addr}; booti ${loadaddr} - ${fdt_addr};'; run mmcboot;
jh_netboot=setenv fdt_file fsl-imx8qxp-mek-root.dtb;setenv boot_os 'scu_rm dtb ${fd                                                                                                          t_addr}; booti ${loadaddr} - ${fdt_addr};'; run netboot;
kboot=booti
loadaddr=0x80280000
loadbootscript=fatload mmc ${mmcdev}:${mmcpart} ${loadaddr} ${script};
loadcntr=fatload mmc ${mmcdev}:${mmcpart} ${cntr_addr} ${cntr_file}
loadfdt=fatload mmc ${mmcdev}:${mmcpart} ${fdt_addr} ${fdt_file}
loadimage=fatload mmc ${mmcdev}:${mmcpart} ${loadaddr} ${image}
loadm4image_0=fatload mmc ${mmcdev}:${mmcpart} ${loadaddr} ${m4_0_image}
m4_0_image=m4_0.bin
m4boot_0=run loadm4image_0; dcache flush; bootaux ${loadaddr} 0
mfgtool_args=setenv bootargs console=${console},${baudrate} rdinit=/linuxrc clk_ign                                                                                                          ore_unused
mmcargs=setenv bootargs console=${console},${baudrate} earlycon root=${mmcroot}
mmcautodetect=yes
mmcboot=echo Booting from mmc ...; run mmcargs; if test ${sec_boot} = yes; then if                                                                                                           run auth_os; then run boot_os; else echo ERR: failed to authenticate; fi; else if t                                                                                                          est ${boot_fdt} = yes || test ${boot_fdt} = try; then if run loadfdt; then run boot                                                                                                          _os; else echo WARN: Cannot load the DT; fi; else echo wait for boot; fi;fi;
mmcdev=0
mmcpart=1
mmcroot=/dev/mmcblk0p2 rootwait rw
netargs=setenv bootargs console=${console},${baudrate} earlycon root=/dev/nfs ip=dh                                                                                                          cp nfsroot=${serverip}:${nfsroot},v3,tcp
netboot=echo Booting from net ...; run netargs;  if test ${ip_dyn} = yes; then sete                                                                                                          nv get_cmd dhcp; else setenv get_cmd tftp; fi; if test ${sec_boot} = yes; then ${ge                                                                                                          t_cmd} ${cntr_addr} ${cntr_file}; if run auth_os; then run boot_os; else echo ERR:                                                                                                           failed to authenticate; fi; else ${get_cmd} ${loadaddr} ${image}; if test ${boot_fd                                                                                                          t} = yes || test ${boot_fdt} = try; then if ${get_cmd} ${fdt_addr} ${fdt_file}; the                                                                                                          n run boot_os; else echo WARN: Cannot load the DT; fi; else booti; fi;fi;
panel=NULL
script=boot.scr
sd_dev=1
sec_boot=no
soc_type=imx8qxp
[... cut ...]
xenlinux_console=hvc0 earlycon=xen
xenmmcboot=setenv get_cmd "fatload mmc ${mmcdev}:${mmcpart}";setenv console ${xenli                                                                                                          nux_console};run mmcargs;run xenboot_common;
xennetboot=setenv get_cmd dhcp;setenv console ${xenlinux_console};run netargs;run x                                                                                                          enboot_common;

Environment size: 4190/8188 bytes
=>

nxpfrankli, 2020-05-14:
Name is wrong.  Update your script,

 23376384 Image-imx8_all.bin -> Image
83922 Image-fsl-imx8qxp-mek.dtb -> fsl-imx8qxp-mek.dtb

pingwuu, 2020-05-16:
@nxpfrankli ,
Yes, now is boot normal after change script. thanks.

closed 2020-05-16 by pingwuu