nxp-imx/mfgtools #190: Help in flashing the i.MX 8M Quad EVK with Linux

francescarover, 2020-06-08:
Hi everyone,
sorry I'm a new user of MX8MQ and in general of this kind of micro. I have to flash Linux on SD card from a widows host PC.
I'using uuu_imx8mq_com_4.14.98 package and this is the result on the command window:

D:\NXP_i_MX_8M\uuu_imx8mq_com_4.14.98>uuu.exe full_tar.uuu
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.154-0-g47ddef5

Your console don't support VT mode, fail back to verbose mode
Build in config:
        Pctl     Chip            Vid     Pid     BcdVersion
        ==================================================
        SDPS:    MX8QXP          0x1fc9  0x012f  [0x0002..0xffff]
        SDPS:    MX8QM           0x1fc9  0x0129  [0x0002..0xffff]
        SDPS:    MX8DXL          0x1fc9  0x0147
        SDPS:    MX28            0x15a2  0x004f
        SDPS:    MX815           0x1fc9  0x013e
        SDPS:    MX865           0x1fc9  0x0146
        SDP:     MX7D            0x15a2  0x0076
        SDP:     MX6Q            0x15a2  0x0054
        SDP:     MX6D            0x15a2  0x0061
        SDP:     MX6SL           0x15a2  0x0063
        SDP:     MX6SX           0x15a2  0x0071
        SDP:     MX6UL           0x15a2  0x007d
        SDP:     MX6ULL          0x15a2  0x0080
        SDP:     MX6SLL          0x1fc9  0x0128
        SDP:     MX7ULP          0x1fc9  0x0126
        SDP:     MXRT106X        0x1fc9  0x0135
        SDP:     MX8MM           0x1fc9  0x0134
        SDP:     MX8MQ           0x1fc9  0x012b
        SDPU:    SPL             0x0525  0xb4a4  [0x0000..0x04ff]
        SDPV:    SPL1            0x0525  0xb4a4  [0x0500..0x9998]
        SDPU:    SPL             0x0525  0xb4a4  [0x9999..0x9999]
        SDPU:    SPL             0x3016  0x1001  [0x0000..0x04ff]
        SDPV:    SPL1            0x3016  0x1001  [0x0500..0x9998]
        FBK:                     0x066f  0x9afe
        FBK:                     0x066f  0x9bff
        FB:                      0x0525  0xa4a5
        FB:                      0x18d1  0x0d02
Wait for Known USB Device Appear...
New USB Device Attached at 1:16
1:16>Start Cmd:SDP: boot -f files/imx-boot-imx8mqea-com-sd.bin
6400%1:16>Okay (0.688s)
New USB Device Attached at 1:16
1:16>Start Cmd:SDP: boot -f files/imx-boot-imx8mqea-com-sd.bin
6400%1:16>Okay (1.295s)

This is what I read on serial COM:

U-Boot SPL 2018.03-2.0.0+gc05d6e576f (Apr 15 2020 - 18:57:07 +0000)
power_bd71837_init
Can't find PMIC:BD71837
DDRINFO: start DRAM init
DRAM PHY training for 3200MTS
check ddr_pmu_train_imem code
check ddr_pmu_train_imem code pass
check ddr_pmu_train_dmem code
check ddr_pmu_train_dmem code pass
Training PASS
DRAM PHY training for 667MTS
check ddr_pmu_train_imem code
check ddr_pmu_train_imem code pass
check ddr_pmu_train_dmem code
check ddr_pmu_train_dmem code pass
Training PASS
DRAM PHY training for 3200MTS
check ddr_pmu_train_imem code
check ddr_pmu_train_imem code pass
check ddr_pmu_train_dmem code
check ddr_pmu_train_dmem code pass
Training PASS
DDRINFO:ddrphy calibration done
DDRINFO: ddrmix config done
Normal Boot
Trying to boot from USB SDP
"Error" handler, esr 0xbf000002
elr: 00000000007e9050 lr : 00000000007e8edc
x0 : 0000000000000000 x1 : 0000000000000000
x2 : 0000000000000387 x3 : 0000000000000380
x4 : 0000000000000010 x5 : 0000000000000000
x6 : 0000000000000070 x7 : 0000000000000388
x8 : 0000000000183c20 x9 : 0000000000000002
x10: 00000000ffffffd0 x11: 0000000000000006
x12: 000000000001869f x13: 0000000000000040
x14: 00000000303100f0 x15: 00000000ffffffff
x16: 0000000000000033 x17: 0000000000000000
x18: 0000000000183e40 x19: 0000000000000387
x20: 0000000000000000 x21: 0000000000008000
x22: 00000000007f6fc7 x23: 0000000000183d40
x24: 00000000007f6faf x25: 0000000000802000
x26: 00000000deadbeef x27: 0000000000000000
x28: 00000000000000f0 x29: 0000000000183c60

Resetting CPU ...

resetting ...
õ

It seems to me a usb problem, for that reason I've dowloaded and installed the winusb driver for WIN7, I've also used ZADIG.exe application to change the driver of from HID to WINusb for SEBlankM850, but the result is the same and I've always have the WCID flasg with a red cross.
I've also tried with Windows10 but the result is the same.
Has anyone of you experienced the same problem?
Thank you to anyone could help me.
Francesca

nxpfrankli, 2020-06-08:
It is your uboot problem.

francescarover, 2020-06-08:
sorry I do not understand, what is my uboot? What do I have to do?

nxpfrankli, 2020-06-08:
Start Cmd:SDP: boot -f files/imx-boot-imx8mqea-com-sd.bin

imx-boot-imx8mqea-com-sd.bin need match your boards.

imx-boot-imx8mqea-com-sd.bin is uboot file, which run at your platform.

francescarover, 2020-06-09:
But if this is not the correct package for the evaluation board of i.MX8M, where can I find it?

nxpfrankli, 2020-06-09:
Did you use NXP published i.MX8MQ EVK boards?
nxp.com have release package.

francescarover, 2020-06-12:
THank you, I've downloaded the latest version LF_v5.4.y-1.0.0_images_MX8MQEVK from the NXP website and it works.
thank you very much.

closed 2020-06-12 by francescarover