nxp-imx/mfgtools #191: Is it possible flash root.tar.xz file to emmc via uuu

pingwuu, 2020-06-10:
Hi,
We can flash root.tar.bz2 to emm via uuuc, but when I flash a root.tar.xz to emmc, it is not success.
So it is possilbe flash root.tar.xz file to emmc via uuu thanks.

jordonwu, 2020-06-11:
> are you use linux kernel to burn root.tar.bz2?
yes,  I using kernel to burn root.tar.bz2 is ok, but burn root.**tar.xz** is failed.
I attached my script
[example_kernel_emmc - Copy.zip](https://github.com/NXPmicro/mfgtools/files/4762394/example_kernel_emmc.-.Copy.zip)

Thanks.

pingwuu, 2020-06-11:
are you use linux kernel to burn root.tar.bz2?
yes, I using kernel to burn root.tar.bz2 is ok, but burn root.tar.xz is failed.
I attached my script
Thanks.
[example_kernel_emmc - Copy.zip](https://github.com/NXPmicro/mfgtools/files/4762405/example_kernel_emmc.-.Copy.zip)

nxpfrankli, 2020-06-11:
you build initramfs by yourself to add xz support.

pingwuu, 2020-06-11:
Get it, I will try later, thanks.

closed 2020-06-12 by pingwuu