nxp-imx/mfgtools #192: Is it possible flash one file or folder (user space application file or folder) to target board's rootfs?

pingwuu, 2020-06-12:
Hi,
I have q question about uuu tool:
1. First, we flash kernel and rootfs to board emmc using uuu kernel flash
2. After this, we need flash a individual file or folder to target board rootfs (such as to board's folder /opt)
Does uuu can do this job? thanks.

nxpfrankli, 2020-06-12:
Do you use uboot or mini kernel to burn rootfs?

pingwuu, 2020-06-12:
we use mini kernel to burn rootfs

nxpfrankli, 2020-06-12:
it should be easy.
Just change uuu script. 

just use ucp cmd to copy to target directory.

pingwuu, 2020-06-12:
@nxpfrankli ,
Is there any sample script for reference? thanks :)

pingwuu, 2020-06-12:
Hi，
I add ucp command as below, but seems not copy a.txt to target.
Any suggestion? thanks.

FBK: ucmd mmc=`cat /tmp/mmcdev`; mkfs.ext3 -F -E nodiscard /dev/mmcblk${mmc}p2
FBK: ucmd mkdir -p /mnt/ext3
FBK: ucmd mmc=`cat /tmp/mmcdev`; mount /dev/mmcblk${mmc}p2 /mnt/ext3

FBK: acmd export EXTRACT_UNSAFE_SYMLINKS=1; tar -jx -C /mnt/ext3
FBK: ucp  core-image-minimal-imx8qxpc0mek.tar.bz2 t:-

**FBK: ucp a.txt t:/usr**

FBK: Sync
FBK: ucmd umount /mnt/ext3

///////////////log output from uuu mini kernel console/////////////////
run shell cmd: export EXTRACT_UNSAFE_SYMLINKS=1; tar -jx -C /mnt/ext3
WOpen:-
WOpen:/usr
WOpen:/usr/a.txt
wait for async proccess finish
run shell cmd: umount /mnt/ext3

nxpfrankli, 2020-06-12:
Put FBK: ucp a.txt t:/usr after Sync.

pingwuu, 2020-06-12:
I will try it next Monday, thanks

pingwuu, 2020-06-15:
> Put FBK: ucp a.txt t:/usr after Sync.
@nxpfrankli ,
I have adjust the ucp cmd after Sync, but a.txt is still not write to target. Would you help check it, thanks.

run shell cmd: mkdir -p /mnt/ext3
run shell cmd: mmc=`cat /tmp/mmcdev`; mount /dev/mmcblk${mmc}p2 /mnt/ext3
[   22.829870] EXT4-fs (mmcblk0p2): mounting ext3 file system using the ext4 subsystem
[   22.842454] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
run shell cmd: export EXTRACT_UNSAFE_SYMLINKS=1; tar -jx -C /mnt/ext3
WOpen:-
wait for async proccess finish
WOpen:/usr/a.txt
run shell cmd: umount /mnt/ext3

///// ls /usr after target reboot
# ls /usr/
aarch64-poky-linux  bin  games  include  lib  libexec  sbin  share  src

nxpfrankli, 2020-06-15:
/usr is initramfs path.

it is should be /mnt/ext3/usr

pingwuu, 2020-06-16:
Yes, now is worked, thanks for your support.

closed 2020-06-16 by pingwuu