nxp-imx/mfgtools #193: "FB:< echo ucmd print"  Example description and successful use?

flobro30101, 2020-06-12:
Has anyone successfully used the shell command? 
I added  simple "FB:< echo ucmd printenv serial#" 

and I get:
1:2>Start Cmd:FB: ucmd printenv serial#
New USB Device Attached at 1:2
1:2>Fail Failure claim interface(0.201s)
1:2>Fail Failure claim interface(0.201s)

Am I missing something?

I would like to use this to get info so I can generate a MAC address for my boards.

Thanks

nxpfrankli, 2020-06-12:
run at windows or linux?
is it success if direct use 
FB: ucmd printenv serial#

flobro30101, 2020-06-12:
Sorry, it is running on Linux but not working...

flobro30101, 2020-06-12:
Also, everything else works in my scripts...

nxpfrankli, 2020-06-12:
I duplicate your problem. let me try to fix it

closed 2020-06-12 by nxpfrankli

flobro30101, 2020-06-12:
Amazing speed.. It works as advertised.

Thanks

nxpfrankli, 2020-06-12:
Can you try https://ci.appveyor.com/project/nxpfrankli/mfgtools-kvqcg/builds/33490219/job/43lwh43mq1dwea6q/artifacts?

flobro30101, 2020-06-12:
I pulled the latest git changes and they worked.
Thanks

Floyd Brown

commit bda58f54 (2020-06-12): fix #193 fb:< echo ucmd print failure

device open twice cause failure