nxp-imx/mfgtools #194: libusb issues during multi-board flashing on Windows 10

ajithpv, 2020-06-16:
Hi,
I'm facing multiple issues w.r.t the _libusb_ during multi-board flashing on Windows 10 PC.
I'm using _uuu version 1.3.102_ and specifying two different USB port/path numbers for 2 i.MX8 platforms.

During my single board trials, I didn't see any issues. However, Multi-board (2 boards) flashing is not reliable with the same set-up and binaries. Sometimes, it will flash without any issues. Other times, it will show _USB_ERROR_PIPE_ for _rootfs_ or larger files. Also, I'm getting _USB_ERROR_TIMEOUT_ rarely. Moreover, the USB flashing will hang after started the flashing without any further prints in some cases (I need to close and start again in this case). 

Hence, I would like to know whether the _libusb_ is reliable on Windows 10 during the multi-board flashing. Whether I need to take care of any know issues in the 2 board flashing time? 

Initially, we were using USB3.0 and then switched to USB 2.0 ports from the PC, still the issues remaining same. It will be difficult to go ahead with these issues from the production point of view.

Could anyone please let me know faced such issues and how to resolve these issues?

nxpfrankli, 2020-06-16:
Did you power MX28 by USB port?

ajithpv, 2020-06-16:
> Did you power MX28 by USB port?  

Sorry, I didn't get what is MX28.

I would like to inform that, we are using a  custom i.MX8 board. Regarding the power, we are using a separate power adapter. The USB cable is connecting from Windows 10 PC to the board directly only for the flashing purpose.

nxpfrankli, 2020-06-16:
Did you use mini linux kernel or uboot to burn Image?

ajithpv, 2020-06-17:
> 
> 
> Did you use mini linux kernel or uboot to burn Image?

I'm using U-boot to burn the images. The default flashing procedure is followed from the NXP i.MX8 board flashing. However, we also have other images (application side) to be copied apart from the U-boot, Kernel and rootfs.

Could you please let me know whether anyone come across this kind of reliability issues earlier? Also, possible solutions to make sure everything works fine too.

nxpfrankli, 2020-06-17:
It passed 4 boards (imx8MM) stress test. I fix some cirtical issue on libusb. 
Maybe you can use below workaround to burn multi board. 

lanuch many uuu instance with -m <port number> option.

ajithpv, 2020-06-18:
I'm already running 2 board parallel flashing with -m option to specify the USB ports. Still this issue is observed. 

May I know the fix for critical issues of libusb applicable to uuu version 1.3.102?

nxpfrankli, 2020-06-18:
https://github.com/libusb/libusb/commits?author=nxpfrankli

ajithpv, 2020-07-10:
> 
> 
> https://github.com/libusb/libusb/commits?author=nxpfrankli

I checked the above link. However, it is not clear to me that, UUU version 1.3.102 is using these libusb fixes.
Could you please confirm that, the 1.3.102 version UUU using these fixes?

nxpfrankli, 2020-07-14:
Yes, it should include the fix

closed 2020-07-15 by ajithpv

Synergetik, 2023-10-11:
Hi @ajithpv,
it's been some years since you encountered the issue described above. But as I'm currently facing pretty much the same problem (see #392), it would be great to learn how you fixed it at the time. 
Did it help to simply switch to another version of the UUU tool + libusb? If so, which versions did work for you?
Thank you very much in advance!