nxp-imx/mfgtools #196: Bulk(R):LIBUSB_ERROR_TIMEOUT

scooty334, 2020-06-28:
Hi guys, im experiencing an issue when i try to download to my i.MX6ULL custom board
zimage, u-boot.imx, imx6ull.dtb, fsl-image-mfgtool-initramfs-imx_mfgtools.cpio.gz.u-boot, works smooth but when it goes to rootfs im getting this error under windows 10 and ubuntu 20.04 also. im using kernel version 4.19.128 if that is matter. 

1:10    21/24 [Bulk(R):LIBUSB_ERROR_TIMEOUT          ] FBK: ucp rootfs.tar t:-

this happens when trying to load rootfs.tar and disconnects at 14%. 

Here is the script: 

==================================================================
# burn rootfs
FBK: ucmd source /tmp/mtd.sh; flash_erase /dev/mtd${nandrootfs} 0 0
FBK: ucmd source /tmp/mtd.sh; ubiattach /dev/ubi_ctrl -m ${nandrootfs}
FBK: ucmd source /tmp/mtd.sh; ubimkvol /dev/ubi0 -Nnandrootfs -m
FBK: ucmd source /tmp/mtd.sh; mkdir -p /mnt/mtd
FBK: ucmd source /tmp/mtd.sh; mount -t ubifs ubi0:nandrootfs /mnt/mtd

FBK: acmd export EXTRACT_UNSAFE_SYMLINKS=1; tar -xv -C /mnt/mtd
FBK: ucp rootfs.tar t:-
FBK: sync
FBK: ucmd umount /mnt/mtd

FBK: done
==================================================================

How can i fix this issue ? 

Thanks

nxpfrankli, 2020-07-14:
It may out of memory,  nand driver get null point from kmalloc and missed a safty check. 

This problem need 5.4 kernel usb driver upgrade, which support scatter list. 

uuu switch to use ffs.

nxpfrankli, 2021-02-11:
if still have problem, please reopen.

closed 2021-02-11 by nxpfrankli