nxp-imx/mfgtools #199: i.MX 8M Nano LIBUSB_ERROR_TIMEOUT

Tonyeeee, 2020-07-06:
Using uuu tool to flashing images from window and ubuntu16.4 giving the same error
#uuu.exe -b emmc_all imx-boot  core-image-minimal-imx8mnevk-20200616160407.rootfs.sdcard.bz2
 
ERROR::   HID(W):LIBUSB_ERROR_TIMEOUT

Can anyone let me know how to debug the issue now ??

nxpfrankli, 2020-07-06:
uuu.exe -b emmc_all <missed bootloader here> imx-boot core-image-minimal-imx8mnevk-20200616160407.rootfs.sdcard.bz2

nxpfrankli, 2020-07-06:
uuu.exe -b emmc_all **missed boot loader here** imx-boot core-image-minimal-imx8mnevk-20200616160407.rootfs.sdcard.bz2

Tonyeeee, 2020-07-07:
thanks nxpfrankli, if we do the the below command then also we got the same error.
uuu.exe -b emmc_all u-boot-imx8mqevk.bin core-image-minimal-imx8mnevk-20200616160407.rootfs.sdcard.bz2
ERROR:: HID(W):LIBUSB_ERROR_TIMEOUT

nxpfrankli, 2020-07-07:
boot loader image should be bootimage, which package spl, uboot \atf together by imx_mkimage

should be imx-boot-imx8mnevk-sd.bin-flash_evk

rohan257, 2020-07-10:
Instead  imx-boot-imx8mnevk-sd.bin-flash_evk, I have imx-boot-imx8mnevk-sd.bin-flash_ddr4_evk
I tried with imx-boot-imx8mnevk-sd.bin-flash_ddr4_evk and still got same error ERROR:: HID(W):LIBUSB_ERROR_TIMEOUT

nxpfrankli, 2020-07-14:
are you using NXP's EVK boards? Any output from uart port? look like DDR init failure

rohan257, 2020-07-15:
Yes, I am using NXP's i.MX8M Nano EVK Board

nxpfrankli, 2020-07-15:
Are you sure you use DDR board, not LPDDR4 board?  there are silk print close to wifi antenna in small CPU boards.

rohan257, 2020-07-15:
Its DDR4, on silk screen print near wifi antenna on CPU board its 8MNANOD4 - EVK written

nxpfrankli, 2020-07-15:
it should okay,  anything out put from serial port? spl should be running. 
try uuu -v -b spl imx-boot-imx8mnevk-sd.bin-flash_ddr4_evk

mb-karo, 2020-09-30:
Having the same issue when trying to program **any** i.MX8M Nano board. No matter if NXP or customboard.

My co-worker spent some time debugging and I might remember it was a problem with device ID.

nxpfrankli, 2021-02-11:
if still have problem, please reopen.

closed 2021-02-11 by nxpfrankli

mb-karo, 2021-02-17:
```
Wait for Known USB Device Appear...
New USB Device Attached at 1:1
1:1>Start Cmd:SDPS: boot -f <imx8m-nano-bootloader>
12%1:1>Fail HID(W):LIBUSB_ERROR_NO_DEVICE(0.109s)
```

Still receiving this error. Then have to restart uuu and then programming begins...

*uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.77-0-g66bf3cc*

nxpfrankli, 2021-02-17:
is it okey, restart board?

mb-karo, 2021-02-18:
If you restart the board, the procedure fails again.
You have to start uuu two times to get it started.

nxpfrankli, 2021-02-18:
Looks like DDR init problem.  first time ddr init failure.  then ROM reset chip, but second time ddr init successed.