nxp-imx/mfgtools #201: Write the documentation in english.

andreacampanella, 2020-07-14:
The Wiki it's wroten in very poor english, the documentation it's unintelligible and that's why so many people have so many problems with this utility, let alone the grammar the rest it's confusing at best: 

"Version show minimize version of uuu to run this script."

minimize? what's that means? should be something like:

"The uuu_version variable holds the minimal version review able to run this script"  

I'm not a native english speaker, I've been in UK for 6 years now, from a company like NXP I expect way better then this. 

I love way too many hours trying to understand what the wiki it's talking about.

nxpfrankli, 2020-07-14:
I am not native engish speaker.  If there are some thing confused, please send suggestion to modified.

github wiki don't support pull -request. 

 "Version show minimize version of uuu to run this script."

means,  if your uuu version below than uuu script declared version number,  uuu will report error and require you update uuu version. 

how about 

"version number represent the minimum uuu version that uuu can parse this script"

andreacampanella, 2020-07-15:
Are you part of NXP?

nxpfrankli, 2020-07-15:
Yes,  but this is independent opensource project.

nxpfrankli, 2020-07-16:
Open source project is difference thing as official endorsed. 
This NXP's official github site.  

Open source project is the method to publish and maintain software. 
Most imx* processor use this tools, but there are also thirdpart tools to support it.

of course All NXP official document point to here.   
open source project can get quicker feedback and quick evolution speed and better quality in long term.  

Only me to maintain this tools now at part time.  What I can do is response all requests in time. 

There already are many global valuable contribitor now. 

 I expect the feedback is specific problem or where what you want to change? like which sentence wrong or confused, what you want to change to.  The general feedback like (documnt is bad,  help missleading) does not help to improve at all.

andreacampanella, 2020-07-16:
I'm still look into it but many debug messages are wrong. 
like : 
Error "Can validate IVT header" 

it's wrong. 

That message it's saying that can check the IVT header? I don't think so. 

It should say "Cannot find valid IVT header" 

There are many many error like this, I just started to look into it but basically most of the line of text into the software are wrong, especially the errors.

andreacampanella, 2020-07-16:
https://github.com/NXPmicro/mfgtools/blob/master/libuuu/cmd.cpp line 667:

Wrong : set_last_err_string("Error: Failed to read the pipe to the end.\n");
Right  : set_last_err_string("Error: Failed to read the end of the pipe.\n");

line 800: 

Wrong :  str = "Current uuu version is too low, please download latest one";
Right  :    str = "This version of uuu version is too old, please download the latest one";

fastboot.cpp : 

set_last_err_string("ucp just support one is remote file start with t:");

I don't even know what this is supposed to mean. 

I only opened random files, you should get the point.

nxpfrankli, 2020-07-16:
Thanks. I fixed previous two error. 
set_last_err_string("ucp just support one is remote file start with t:");

means only one (source or destination) support "t:" 

ucp <source path>  t:<destination>   from PC to boards
ucp t:<source path> <destination>  from boards to PC.

ajithpv, 2020-08-15:
@andreacampanella  - Considering the amount of work which @nxpfrankli is doing himself to support on the UUU tool, we would appreciate him and make it better by our contribution.
Please feel free to add your contributions to the documentation which obliviously will help many people who are using it.

@nxpfrankli - I would like to know your preferred way for the document contributions. Everyone can follow that and make the documentation better.

nxpfrankli, 2020-08-16:
if in code, please send pull request to me.
if in wiki page, you can clone wiki page repo and send patch to me because github can't support pull request for wiki page.

closed 2021-02-11 by nxpfrankli