nxp-imx/mfgtools #208: I.MX8QXP  Fail HID(R):LIBUSB_ERROR_TIMEOUT

hey-monster, 2020-08-04:
Hello, im experiencing an issue when i try to download to my i.MX8QX custom board.

i use imx-mkimage compile the flash.bin success, using(scfw, u-boot, atf, ahab) . 
Then i use uuu to load the  imx-boot-imx8qxpmek-sd.bin-flash, commond is "uuu.exe -b emmc_all imx-boot-imx8qxpmek-sd.bin-flash fsl-image-validation-imx-imx8qxpmek-20200713031137.rootfs.sdcard".
when it goes to the first imx-boot-imx8qxpmek-sd.bin-flash im getting this error under windows 10. 
im using kernel version 4.14.98.

1:8    1/ 1   [HID(R):LIBUSB_ERROR_TIMEOUT ] SDPS: boot -f imx-boot-imx8qxpmek-sd.bin-flash

this happens when trying to load bin-flash and disconnects at 19%, and the serial do not have any logs.
when i excute commond "uuu.exe -v -b spl imx-boot-imx8qxpmek-sd.bin-flash"
it shows:
# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f imx-boot-imx8qxpmek-sd.bin-flash

# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM
SDPS: boot -f imx-boot-imx8qxpmek-sd.bin-flash

# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
# {
SDPU: delay 1000
SDPU: write -f imx-boot-imx8qxpmek-sd.bin-flash -offset 0x57c00
SDPU: jump
SDPU: done
# }

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f imx-boot-imx8qxpmek-sd.bin-flash -skipspl
SDPV: jump
SDPV: done
# }

Wait for Known USB Device Appear...
New USB Device Attached at 1:8
1:8>Start Cmd:SDPS: boot -f imx-boot-imx8qxpmek-sd.bin-flash
19%1:8>Fail HID(R):LIBUSB_ERROR_TIMEOUT

Can anyone help me ? I have struggled for weeks

nxpfrankli, 2020-08-04:
Most case is DDR init failure by SCU firmware.

nxpfrankli, 2021-02-11:
if still have problem, please reopen.

closed 2021-02-11 by nxpfrankli