nxp-imx/mfgtools #212: error MSB3030: Could not copy the file $(SolutionDir)c.txt

keelung-yang, 2020-08-13:
on latest master, both in `uuu.vcxproj` and `uuu-static-link.vcxproj`
```shell
  Generating Code...
  uuu.vcxproj -> C:\dev\mfgtools\msvc\..\x64\Debug\dll\uuu.exe
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targets(4651,5): error MSB3030: Could not copy the file "C:\dev\mfgtools\msvc\c.txt" because it was not found.
```
And `$(SolutionDir)\c.txt` in vcxproj should be `$(SolutionDir)c.txt` if needed indeed.

And what's the meaning of **_c_**.txt?

nxpfrankli, 2020-08-13:
it should be for early phase debug and forget remove it.
I just removed it.

keelung-yang, 2020-08-14:
> it should be for early phase debug and forget remove it.
> I just removed it.

Then it shouldn't be committed.

nxpfrankli, 2020-08-14:
You are right. but it is in vcproj file, which genenally is not human read file. 
VS GUI don't show this pre-action.

keelung-yang, 2020-08-14:
It's human readable, and VS GUI shown it: 

[image]

nxpfrankli, 2020-08-14:
Thank you for point out. 
I just checked pre-build event.  

Anyway I already removed it. 

it is no issue about double \\.  OS can handle it correctly.  Our CI also have not reported any problem.

closed 2020-08-14 by keelung-yang