nxp-imx/mfgtools #213: flash -raw2sparse

bevincb, 2020-08-13:
Please let us know the use of "-raw2sparse" option in flash command of uuu tool.

Example:
FB: flash -raw2sparse all k-base-image-ma035-01.wic.bz2\*

Can you please help us to understand what will happen on execution of the above command.

nxpfrankli, 2020-08-13:
uboot only accept android sparse file format for **all** partion. -raw2sparse means convert binary raw file into sparse format run time.

nxpfrankli, 2021-02-11:
if still have problem, please reopen.

closed 2021-02-11 by nxpfrankli

nxpfrankli, 2023-05-17:
How much reduced?  raw2spare using the simple method.  In our release built image, there are few sparse area exist. especially using zstd or other compress method.  Sparse information may lost during compression.