nxp-imx/mfgtools #214: Wrong UUU output path in uuu(-static-link).vcxproj

keelung-yang, 2020-08-14:
Since uuu is an application, not a static or dynamic library, it shouldn't be outputted to a lib/dll directory.
It's better to put it into the same directory as other vc projects in .sln:

[image]

[image]

nxpfrankli, 2020-08-14:
Yes, But I don't want to change libusb directory, which submodule to libusb git repo. 
So I follow up libusb output directory. 

At beginning, we use dynatmic link because libusb is lgpl license.   If uuu put to other directory, uuu can't be launch by VS because missed libusb.dll. 

After clearfy we can static link libusb, we continue to use this method.

keelung-yang, 2020-08-14:
uuu can't be launched now without modifying. 
[image]
Since uuu.exe in `mfgtools\x64\Debug\dll` and libusb-1.0.dll in `mfgtools\libusb\x64\Debug\dll`

No need to modify libusb, just add a post-build event to uuu project settings, such as
```copy /y "$(SolutionDir)..\libusb\$(Platform)\$(Configuration)\dll\*.dll"  $(Platform)\$(Configuration)\```

nxpfrankli, 2020-08-14:
I have not touch dynamtic link for long time. Maybe libusb change it

closed 2020-08-14 by nxpfrankli

keelung-yang, 2020-08-14:
What kind of touching as a reason of refusing to fix?
**uuu can't be launched by vs now without modifying**, IS  THAT OK?

nxpfrankli, 2020-08-14:
fixed patch pushed. System auto close it.

commit 9aff92a8 (2020-08-14): fixed #214: Wrong output path in uuu(-static-link).vcxproj

move output uuu.exe under sln folder
copy libusb.dll to the same dir for uuu.sln