nxp-imx/mfgtools #216: i.MX8QXP: support for container with SPL

niebelm, 2020-08-28:
Current Version does not support Container containing U-Boot SPL + second container with ATF + U-Boot. Would be nice to have support for this use case.

Currently only the first AP payload (SPL in our case) is downloaded and started. Using SPDV commands to download flash.bin to the running SPL is possible, but then the system hangs.

nxpfrankli, 2020-08-28:
Already supported.   try latest nxp released uboot

niebelm, 2020-08-31:
Thanks. Reason was a twofold error:

1. uuu is not able to split flash.bin container generated with flash_spl_flexspi. Error from uuu "Unknown Image type, can't use skipspl" was misleading
2. our U-Boot config had a problem with env when booting from USB which made our U-Boot silently fail

closed 2020-08-31 by niebelm