nxp-imx/mfgtools #218: uuu@: IOUSBUserClientLegacy::start: missing entitlement com.apple.ibridge.control with uuu on Mac

michaellihs, 2020-08-28:
When I try to flash a (proprietary) hardware device from my Mac (macOS Mojave 10.14.6), I get the following error message in the terminal:

```
$ sudo uuu xyz.uuu
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.43-4-g9aff92a

Success 0    Failure 1

20:32    1/ 0 [Failure open usb device,Try sudo uuu  ]
```

After reading some posts around this issue I was made aware of checking `dmesg` entries and see the following log:

```
uuu@: IOUSBUserClientLegacy::start: missing entitlement com.apple.ibridge.control
```

Is that a know issue when using `uuu` on a Mac or something I did wrong? Can I somehow work around this / fix this?

Before using `uuu`, I run the following script to build it on my Mac (that should show dependencies and 3rd party libs used):

```bash
#!/bin/sh

git clone https://github.com/NXPmicro/mfgtools.git
brew install cmake openssl pkg-config libusb libzip

sudo ln -s /usr/local/Cellar/openssl@1.1/1.1.1g/lib/libcrypto.1.1.dylib /usr/local/lib/libcrypto.dylib
sudo ln -s /usr/local/Cellar/openssl@1.1/1.1.1g/lib/libssl.1.1.dylib /usr/local/lib/libssl.dylib

cd mfgtools
cmake -DOPENSSL_CRYPTO_LIBRARY="/usr/local/lib/libcrypto.dylib" -DOPENSSL_SSL_LIBRARY=/usr/local/lib/libssl.dylib .
make
```

Thanks for any help!

nxpfrankli, 2020-08-29:
it need unload HID driver. but it is fail at latest MAC OS. It need a sigature to unload driver. but we still not get sigature from apple yet.   
kextunload -b com.apple.driver.usb.IOUSBHostHIDDevice

michaellihs, 2020-08-29:
Hm - I assume, since 

```
kextunload -b com.apple.driver.usb.IOUSBHostHIDDevice
```

returns an error message, namely

```
(kernel) Can't remove kext com.apple.driver.usb.IOUSBHostHIDDevice; services failed to terminate - 0xdc008018.
Failed to unload com.apple.driver.usb.IOUSBHostHIDDevice - (libkern/kext) kext is in use or retained (cannot unload).
```

I cannot unload it and have to wait for you to get the signature from Apple and change this in the code - right?

nxpfrankli, 2020-08-31:
That's what I get so far. 
May you look for the other solution.