nxp-imx/mfgtools #221: Intent to package: Snaps are universal Linux packages

Lin-Buo-Ren, 2020-09-14:
Dear maintainer(s), 

[Snaps are universal Linux packages](https://snapcraft.io) that can be installed and run on many GNU+Linux distributions with ease.  I've been following the [Snapcrafters](https://forum.snapcraft.io/t/join-snapcrafters/1325) instructions and [have built a working snap on the Snap Store](https://snapcraft.io/universal-update-utility), you may try it by running the following commands after [installing snapd](https://docs.snapcraft.io/installing-snapd/6735):
 
```
sudo snap install --channel=beta universal-update-utility
sudo snap connect universal-update-utility raw-usb
universal-update-utility -H
# sudo snap alias universal-update-utility uuu
# uuu -H
```

I'm also looking forward to working with you to make the packaging work upstream (if you're interested).  You may want to check out the current [packaging recipe](https://github.com/Lin-Buo-Ren/uuu-snap), the [documentation](https://docs.snapcraft.io), registering a developer account on the [Snap Store](https://snapcraft.io/store), and [get in touch with us](https://forum.snapcraft.io) to transfer the work under your control.

Thanks for your time and attention.

nxpfrankli, 2021-03-02:
Thank you very much.  It think it may not be too late.  I just realize it is quite good features.
I just start learn snap.

Lin-Buo-Ren, 2021-03-02:
I can prepare a PR to add the packaging recipe to the repo and initiate the ownership transfer process.

nxpfrankli, 2021-03-02:
Thanks. I requested the new name "uuu" and wait for review.  universal-update-utility is too long.

Lin-Buo-Ren, 2021-03-02:
> I requested the new name "uuu" and wait for review. universal-update-utility is too long.

I've previously attempted to register 3-letter snap names for other snaps but was since been retracted with the following staff recommendation:

> We're trying to limit the amount of 3-letter snap names because they are not very descriptive unless they're very high-profile software (git, vim). Would you consider using a longer more descriptive name for the snap? as you know we can still assign an alias so a "bfg" command will be available once installed, but naming the snap "big-fantastic-gun" or something more descriptive is preferred.

I assume `uuu` falls under this case and does not attempt it in the first place.

nxpfrankli, 2021-03-02:
Thank you for your comments.

nxpfrankli, 2021-03-03:
Feedback: 

Three-letter and shorter snap names are too generic and nondescript and don't provide a good idea of what the snap actually is or does. For that reason, we prefer and suggest you pick a more descriptive name for your snap (universal-update-utility-uuu for example). Keep in mind that even if the snap has one of those names, a command called "uuu" can still be provided by the snap - this needs creating an "alias" but it's entirely feasible to do so.

Let me know if some of the suggested names are OK with you or if you want to pick a new one; in any case, you can simply register the new name and it should be automatically approved - only 3-letter or shorter names (well and a few other exceptions) require manual name review/approval."

Maybe we, can use universal-update-utility-uuu, or uuu-universal-update-utility, so user can use keywords uuu to search it.

Lin-Buo-Ren, 2021-03-04:
> Maybe we, can use universal-update-utility-uuu, or uuu-universal-update-utility, so user can use keywords uuu to search it

I've added search keywords to the `universal-update-utility` snap's listing page and can verify that the snap is now searchable from the store website and the CLI, does that work for you?

[image]

```
brlin@brlin-musicha-mz-530:~$ snap find uuu
Name                      Version          Publisher  Notes  Summary
universal-update-utility  1.4.95+pkg-021c  brlin      -      Freescale/NXP I.MX Chip image deploy tools
brlin@brlin-musicha-mz-530:~$ snap find nxp
Name                                    Version          Publisher    Notes  Summary
zephyr-crosstool-xtensa-nxp-imx-adsp    1.24.0.4         stephanosio  -      Zephyr Xtensa NXP i.MX ADSP Cross Compiler Toolchain
zephyr-crosstool-xtensa-nxp-imx8m-adsp  1.24.0.4         stephanosio  -      Zephyr Xtensa NXP i.MX 8M ADSP Cross Compiler Toolchain
universal-update-utility                1.4.95+pkg-021c  brlin        -      Freescale/NXP I.MX Chip image deploy tools
```

Also, I would like to request to add your Ubuntu One account e-mail address to the `universal-update-utility` snap's collaborator list so that you can access the current snap's backstage.

[image]

Lin-Buo-Ren, 2021-03-04:
I also [fixed the build failure](https://github.com/Lin-Buo-Ren/uuu-snap/commit/021cfbdcbf7b7e4b19c8247148e19dc0c67c0393) and update the current snap to 1.4.95

nxpfrankli, 2021-03-04:
My account is frank.li@nxp.com

Lin-Buo-Ren, 2021-03-04:
Collaborator invited, please check your mail.

Lin-Buo-Ren, 2021-03-04:
I've also added your GitHub account to the source repo's collaborator list, you may push code changes and trigger the automated snap building.

nxpfrankli, 2021-03-04:
I can push code into your repo.  
How to trigger snap build if change source in uuu repo?

Lin-Buo-Ren, 2021-03-04:
> How to trigger snap build if change source in uuu repo?

This is currently not possible if the snap recipe is in another repo, we can work around it by:

1. Merging the snap recipe to the uuu repo.
1. Transfer the snap's ownership to your developer account (probably not required with contributor access)
1. Setup snap building with the uuu repo at <https://snapcraft.io/universal-update-utility/builds>.

nxpfrankli, 2021-03-04:
404 for link https://snapcraft.io/universal-update-utility/builds.

nxpfrankli, 2021-03-04:
About 1.  Just need copy everything under snap into uuu repo?

Lin-Buo-Ren, 2021-03-04:
> 404 for link https://snapcraft.io/universal-update-utility/builds.

Probably no permission as a snap collaborator, then the ownership transfer process must be done.  Here is the process:

1. Register an account on the snapcraft forum
1. [Post a snap name transfer request under the `store` topic category of the Snapcraft Forum](https://forum.snapcraft.io/search?q=transfer%20category%3Astore), you'll have to mention either your snap account id listed in [https://dashboard.snapcraft.io/dev/account/](https://dashboard.snapcraft.io/dev/account/) or the associated e-mail address. You may check out this example: [Transfer brook - store - snapcraft.io](https://forum.snapcraft.io/t/transfer-brook/4966)  
    I can also make the post for you, in this case simply mention the aforementioned account details here.
1. After the name transfer is completed, the build page should be accessible.

> About 1. Just need copy everything under snap into uuu repo?

I've filed a PR (#255) that adds the necessary files.

Lin-Buo-Ren, 2021-03-04:
I've found more issues while using the uuu snap:

* If the user has no write access to the target device node, one has to run the `uuu` snap command as root ([the `raw-usb` interface](https://snapcraft.io/docs/raw-usb-interface) connection is, unfortunately, not enough)
* If the `uuu` snap command is run as root then due to the current implementation of the `home` interface's security confinement, the program can't access files under the user home directory (i.e. uuu scripts can't be read)

Refer [Libusb application can't access device even after the raw-usb interface is connected - snapcraft - snapcraft.io](https://forum.snapcraft.io/t/libusb-application-cant-access-device-even-after-the-raw-usb-interface-is-connected/23069) for more info.

A dirty workaround is to make the user modify the permission of the target device mode manually(0666) and run `uuu` as the user instead of root.

nxpfrankli, 2021-03-04:
pull request merged. 

I created snapcraft account, email is frank.li@nxp.com. please help post transfer request for me

About sudo access usb device problem. 

1: put above udev run into /etc/udev/rules.d/70-uuu.rules
        sudo sh -c "uuu -udev >> /etc/udev/rules.d/70-uuu.rules"
2: update udev rule
        sudo udevadm control --reload

I am not sure how to integrate into snap

Lin-Buo-Ren, 2021-03-04:
> pull request merged.

~~Please merge #256 which fixes errors in the original PR.~~

> I created snapcraft account, email is frank.li@nxp.com. please help post transfer request for me

Please provide the snap account id listed in <https://dashboard.snapcraft.io/dev/account/>.

> I am not sure how to integrate into snap

Unfortunately, it doesn't seem to be supported as of now.  You might want to add the instructions to the snap's launcher <https://github.com/NXPmicro/mfgtools/blob/master/snap/local/launchers/universal-update-utility-launch> so that the user will be notified of the workaround.

nxpfrankli, 2021-03-04:
Snap account-id
NxFU2gRxjZ0LR5Ma7eNl05ZapIZtsxoK

Lin-Buo-Ren, 2021-03-04:
Thanks!  The snap ownership transfer request is filed at <https://forum.snapcraft.io/t/please-transfer-universal-update-utility-to-nxpfrankli/23120>

nxpfrankli, 2021-03-04:
Our license team require add your MIT license file under snap

Lin-Buo-Ren, 2021-03-04:
> Our license team require add your MIT license file under snap

I filed a new PR (#257) which should meet the requirement.

nxpfrankli, 2021-03-04:
Thank very much

nxpfrankli, 2021-03-04:
Sorry, license team review and find one more thing. 

Please add copyright at yaml file

Copyright (c) 2017 Snapcrafters

nxpfrankli, 2021-03-04:
One more question: 
how to distinguish beta and stable package?  Most people should get stable package default.

Lin-Buo-Ren, 2021-03-04:
> Please add copyright at yaml file

Done. (#258)

> how to distinguish beta and stable package? Most people should get stable package default.

The snap ecosystem features [channels and tracks](https://snapcraft.io/docs/channels) for this purpose.  The new build will always land in the `edge` channel and can be _promoted_ to other channels via store backstage or the `snapcraft promote` command.

```
channels:
  latest/stable:    –                                           
  latest/candidate: –                                           
  latest/beta:      1.4.95+pkg-021c         2021-03-02 (37) 1MB -
  latest/edge:      1.4.95+pkg-021c         2021-03-02 (37) 1MB -
```

This snap implementation also integrates a selective-checkout script which, will help ensure that stable/candidate/beta snap revision will always be built for promotion, however, it relies on distinguishable naming of git tags so it doesn't for uuu.  Refer [Selective-checkout: Check out the tagged release revision if it isn't promoted to the stable channel - doc - snapcraft.io](https://forum.snapcraft.io/t/selective-checkout-check-out-the-tagged-release-revision-if-it-isnt-promoted-to-the-stable-channel/10617) for more details.

Lin-Buo-Ren, 2021-03-04:
> Also, I would like to request to add your Ubuntu One account e-mail address to the universal-update-utility snap's collaborator list so that you can access the current snap's backstage.

Please accept the snap collaboration invite in order to proceed with the ownership transfer [according to the store staffs' request](https://forum.snapcraft.io/t/please-transfer-universal-update-utility-to-nxpfrankli/23120/2).

[image]

Lin-Buo-Ren, 2021-03-04:
> I've found more issues while using the uuu snap:
>     * If the `uuu` snap command is run as root then due to the current implementation of [the `home` interface](https://snapcraft.io/docs/home-interface)'s security confinement, the program can't access files under the user home directory (i.e. uuu scripts can't be read)

I found that this is no longer the case as the `home` interface supports a `read` attribute which, when set to `all`, also allows reading non-hidden files in the home directories of all users as traditional file permissions allow.  However, setting this attribute will make the interface non-auto-connect and requires a manual interface connection(`sudo snap connect universal-update-utility:home`) or a store approval for autoconnect by default.  Refer [Process for aliases, auto-connections and tracks - store-requests - snapcraft.io](https://forum.snapcraft.io/t/process-for-aliases-auto-connections-and-tracks/455) for details.

I've initiated the interface auto-connect request: [Interface auto-connect request for the universal-update-utility snap - store-requests - snapcraft.io](https://forum.snapcraft.io/t/interface-auto-connect-request-for-the-universal-update-utility-snap/23125)

nxpfrankli, 2021-03-04:
accepted invite

Lin-Buo-Ren, 2021-03-04:
> accepted invite

I've updated the request's status.

nxpfrankli, 2021-03-04:
[image]

I see.  https://snapcraft.io/universal-update-utility/builds

I have not seen edit button.

Lin-Buo-Ren, 2021-03-05:
> I have not seen edit button

To set up a new repo for snap building, disconnect the old repo and connect to the new one.

nxpfrankli, 2021-03-05:
After I update selective-checkout part, get below error. 

Staging selective-checkout 
+ snapcraftctl stage
Sorry, an error occurred in Snapcraft:
[Errno 2] No such file or directory: '/build/universal-update-utility/parts/selective-checkout/install/scriptlets'
Traceback (most recent call last):
  File "/snap/snapcraft/6120/bin/snapcraft", line 33, in <module>
    sys.exit(load_entry_point('snapcraft==4.5.4', 'console_scripts', 'snapcraft')())
  File "/snap/snapcraft/6120/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/snap/snapcraft/6120/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/snap/snapcraft/6120/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/snap/snapcraft/6120/lib/python3.6/site-packages/snapcraft/cli/_command.py", line 89, in invoke
    return super().invoke(ctx)

Lin-Buo-Ren, 2021-03-05:
> After I update selective-checkout part, get below error.

Please revert 6cffef2 as the part isn't meant to be used like that.

>> This snap implementation also integrates a selective-checkout script which, will help ensure that stable/candidate/beta snap revision will always be built for promotion, however, **it relies on distinguishable naming of git tags so it doesn't work for uuu**. Refer [Selective-checkout: Check out the tagged release revision if it isn't promoted to the stable channel - doc - snapcraft.io](https://forum.snapcraft.io/t/selective-checkout-check-out-the-tagged-release-revision-if-it-isnt-promoted-to-the-stable-channel/10617) for more details.

Currently, the part will not do much else than always making the latest tagged release built at least once.

nxpfrankli, 2021-03-05:
reverted. Do you why there are dirty at package name universal-update-utility_1.4.107-1-g8f92a02-**dirty_**s390x.snap?

Lin-Buo-Ren, 2021-03-05:
> Do you why there are dirty at package name universal-update-utility_1.4.107-1-g8f92a02-**dirty_**s390x.snap?

The selective-checkout script somehow determines the repo is modified, I'll check it out.

Lin-Buo-Ren, 2021-03-06:
> Do you why there are dirty at package name universal-update-utility_1.4.107-1-g8f92a02-**dirty_**s390x.snap?

I unable to reproduce the problem after fixing the selective-checkout part (#259), please verify.

nxpfrankli, 2021-03-06:
Still have **dirty** v1.4.107-7-g1b9a4d1-dirty (66) 
https://snapcraft.io/universal-update-utility/releases

Lin-Buo-Ren, 2021-03-06:
> Still have **dirty** v1.4.107-7-g1b9a4d1-dirty (66)

I still unable to reproduce the problem locally:

```build output
Pulling main 
+ set -o nounset
+ snapcraftctl pull
+ /root/stage/scriptlets/selective-checkout --release-tag-pattern=uuu_[[:digit:]]+(\.[[:digit:]]+){2} --release-tag-prefix=uuu_
selective-checkout: The last tagged stable release(1.4.107) hasn't been promoted to the stable channel(1.4.107-1-g8f92a02-dirty) on the Snap Store yet, checking out 1.4.107.
Note: checking out 'uuu_1.4.107'.
```

However, after examining the build log (<https://snapcraft.io/universal-update-utility/builds/1323496>) I found it was the `libusb` submodule clone failure that causes the unclean state of the repo(#252):

```build log
Cloning into 'libusb'...
[06/Mar/2021:14:57:29 +0000] "CONNECT github.com:443 HTTP/1.1" 200 4080981 "-" "git/2.7.4"
[06/Mar/2021:14:57:29 +0000] "CONNECT github.com:443 HTTP/1.1" 200 27358 "-" "git/2.7.4"
fatal: reference is not a tree: e782eeb2514266f6738e242cdcb18e3ae1ed06fa
Cloning into 'zlib'...
[06/Mar/2021:14:57:30 +0000] "CONNECT github.com:443 HTTP/1.1" 200 3947446 "-" "git/2.7.4"
Submodule path 'zlib': checked out 'cacf7f1d4e3d44d871b605da3b647f07d718623f'
Unable to checkout 'e782eeb2514266f6738e242cdcb18e3ae1ed06fa' in submodule path 'libusb'
'git submodule update --init --recursive failed with exit code 1 (build may fail later)
```

```
snapcraft-universal-update-utility ../parts/main/src# git status
On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)

        modified:   libusb (new commits, modified content)

no changes added to commit (use "git add" and/or "git commit -a")

```

It appears to be [a deficiency/bug in Git <2.8 that makes Git unable to check-out the submodule hash](https://stackoverflow.com/a/17692710/6219078):

git 2.7.4(ubuntu 16.04) - failure
git 2.17.1 (ubuntu 18.04) - success
git 2.25.1 (ubuntu 20.04) - success
git 2.27.0 - success

Possible solution:

* Upgrade Git before checking out submodule
* Migrate to a recent Ubuntu Core base(`core18`) that ships a recent Git distribution

Lin-Buo-Ren, 2021-03-06:
> Still have **dirty** v1.4.107-7-g1b9a4d1-dirty (66)

#262 PR should fix the problem.

closed 2021-03-09 by nxpfrankli

Lin-Buo-Ren, 2021-03-21:
> I found that this is no longer the case as [the `home` interface supports a `read` attribute](https://snapcraft.io/docs/home-interface) which, when set to `all`, also allows reading non-hidden files [...]

Due to the snap ownership change and in order to accelerate the review process, I would like to ask for an additional confirmation that I am requesting the new interface assertion on behalf of you.

nxpfrankli, 2021-03-22:
done