nxp-imx/mfgtools #222: IMX8QXP and UUU - Android flash error: Partition does not exist

B-Andras, 2020-09-15:
Dear NXP team,

I would like to ask for your help regarding flashing Android on i.MX 8QuadXPlus MEK board.

I tried to flash android on the board, following every step according to the documentation.
The flashing was tried for Android 10.0.0-2.2.0 AUTO, Android 9.0.0-2.3.2, Android 9.0.0-2.0.0 versions also, with Linux and Windows as well and also from different computers and with many different UUU versions, but none them works.

After the flashing starts it stucks and fails with the following error message:

> 2:2   10/25   [partition does not exist    ] FB: flash bootloader0 u-boot-imx8qxp.imx.link

We have board from TechNexion as well (EDM-IMX8M, imx8qm chip) and I was able to flash Android without problem.

Please let me know if your have any advice how to solve this problem.
Thank you very much for the help in advance.

a09bdd, 2020-09-15:
in the host log, it seems that "bootloader0" partition cannot be recognized by uboot running on the board when flash images.

"bootloader0" partition is a partition hardcoded in uboot for imx8m/imx8 devices. have you ever modified the uboot code?

when uboot ecountered a unrecoginzed partition, it outputs the partitions can be recognized on the board console, would you reproduce this issue and provide that console log to let us check what partitions can now be recognized by uboot running on the board?

B-Andras, 2020-09-15:
Thank you very much for the quick response.

No, I have not modified the uboot code.

I have reproduced the error, please see the log below (It is a Windows PowerShell ISE log)

> PS D:\___560023_NGDP\_dev___Android_imageFiles\android_p9.0.0_2.0.0-ga_image_8qmek> .\uuu_imx_android_flash.bat -f imx8qxp -e
> This script is validated with uuu 1.2.91 version, please align with this version.
> dtbo is supported
> dual slot is supported
> generate lines to flash partition-table.img to the partition of gpt
> generate lines to flash u-boot-imx8qxp.imx to the partition of bootloader0
> generate lines to flash dtbo-imx8qxp.img to the partition of dtbo_a
> generate lines to flash boot.img to the partition of boot_a
> generate lines to flash system.img to the partition of system_a
> generate lines to flash vendor.img to the partition of vendor_a
> generate lines to flash vbmeta-imx8qxp.img to the partition of vbmeta_a
> generate lines to flash dtbo-imx8qxp.img to the partition of dtbo_b
> generate lines to flash boot.img to the partition of boot_b
> generate lines to flash system.img to the partition of system_b
> generate lines to flash vendor.img to the partition of vendor_b
> generate lines to flash vbmeta-imx8qxp.img to the partition of vbmeta_b
> uuu script generated, start to invoke uuu with the generated uuu script
> uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.91-0-g3799f4d
> 
> Your console don't support VT mode, fail back to verbose mode
> Build in config:
> 	Pctl	 Chip		 Vid	 Pid	 BcdVersion
> 	==================================================
> 	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
> 	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
> 	SDP:	 MX7D		 0x15a2	 0x0076
> 	SDP:	 MX6Q		 0x15a2	 0x0054
> 	SDP:	 MX6D		 0x15a2	 0x0061
> 	SDP:	 MX6SL		 0x15a2	 0x0063
> 	SDP:	 MX6SX		 0x15a2	 0x0071
> 	SDP:	 MX6UL		 0x15a2	 0x007d
> 	SDP:	 MX6ULL		 0x15a2	 0x0080
> 	SDP:	 MX6SLL		 0x1fc9	 0x0128
> 	SDP:	 MX7ULP		 0x1fc9	 0x0126
> 	SDP:	 MXRT106X	 0x1fc9	 0x0135
> 	SDP:	 MX8MM		 0x1fc9	 0x0134
> 	SDP:	 MX8MQ		 0x1fc9	 0x012b
> 	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
> 	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
> 	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
> 	FBK:	 		 0x066f	 0x9afe
> 	FBK:	 		 0x066f	 0x9bff
> 	FB:	 		 0x0525	 0xa4a5
> 	FB:	 		 0x18d1	 0x0d02
> Wait for Known USB Device Appear...
> New USB Device Attached at 2:2
> 2:2>Start Cmd:FB: ucmd setenv fastboot_dev mmc 
> 2:2>Okay
> 2:2>Start Cmd:FB: ucmd setenv mmcdev 0 
> 2:2>Okay
> 2:2>Start Cmd:FB: ucmd mmc dev 0 
> 2:2>Okay
> 2:2>Start Cmd:FB: ucmd mmc dev 0 0 
> 2:2>Okay
> 2:2>Start Cmd:FB: ucmd mmc erase 0x2000 0x10 
> 2:2>Okay
> 2:2>Start Cmd:FB: ucmd mmc partconf 0 1 1 1 
> 2:2>Okay
> 2:2>Start Cmd:FB: flash gpt partition-table.img.link 
> 0x400000000x2002:2>Okay
> 2:2>Start Cmd:FB: ucmd setenv fastboot_dev sata 
> 2:2>Okay
> 2:2>Start Cmd:FB: ucmd setenv fastboot_dev mmc 
> 2:2>Okay
> 2:2>Start Cmd:FB: flash bootloader0 u-boot-imx8qxp.imx.link 
> 0x400000000x200partition does not exist2:2>Fail partition does not exist
> [?25h
> 
> 
> 
> PS D:\___560023_NGDP\_dev___Android_imageFiles\android_p9.0.0_2.0.0-ga_image_8qmek> 

In Windows PowerShell (Not ISE version) the log is the following:

> PS D:\___560023_NGDP\_dev___Android_imageFiles\android_p9.0.0_2.0.0-ga_image_8qmek> .\uuu_imx_android_flash.bat -f imx8qxp -e
> This script is validated with uuu 1.2.91 version, please align with this version.
> dtbo is supported
> dual slot is supported
> generate lines to flash partition-table.img to the partition of gpt
> generate lines to flash u-boot-imx8qxp.imx to the partition of bootloader0
> generate lines to flash dtbo-imx8qxp.img to the partition of dtbo_a
> generate lines to flash boot.img to the partition of boot_a
> generate lines to flash system.img to the partition of system_a
> generate lines to flash vendor.img to the partition of vendor_a
> generate lines to flash vbmeta-imx8qxp.img to the partition of vbmeta_a
> generate lines to flash dtbo-imx8qxp.img to the partition of dtbo_b
> generate lines to flash boot.img to the partition of boot_b
> generate lines to flash system.img to the partition of system_b
> generate lines to flash vendor.img to the partition of vendor_b
> generate lines to flash vbmeta-imx8qxp.img to the partition of vbmeta_b
> uuu script generated, start to invoke uuu with the generated uuu script
> uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.91-0-g3799f4d
> 
> Success 0    Failure 1
> 
> 2:2   10/25   [partition does not exist    ] FB: flash bootloader0 u-boot-imx8qxp.imx.link
> PS D:\___560023_NGDP\_dev___Android_imageFiles\android_p9.0.0_2.0.0-ga_image_8qmek>

I would like to note that I have tried every newer UUU version after 1.2.91 also, but I have the same error message.

a09bdd, 2020-09-15:
Hi, @B-Andras, this is the log on the host. would you provide the log of board console, the log is output from USB DEBUG port, it's an usb port besides the ethernet port.

For example, if I tried to flash an image to the partition "fake_partition", which does not exist with the release image, uboot cannot recognize the partition and will output all the partitions it can recognize as below:

`..........
downloading of 1355776 bytes finished
idx 0, ptn 0 name='gpt' start=0 len=2048
idx 1, ptn 0 name='' start=0 len=0
idx 2, ptn 0 name='all' start=0 len=61079552
idx 3, ptn 0 name='bootloader0' start=0 len=8192
idx 4, ptn 1 name='dtbo_a' start=16384 len=8192
idx 5, ptn 2 name='dtbo_b' start=24576 len=8192
idx 6, ptn 3 name='boot_a' start=32768 len=98304
idx 7, ptn 4 name='boot_b' start=131072 len=98304
idx 8, ptn 5 name='system_a' start=229376 len=3670016
idx 9, ptn 6 name='system_b' start=3899392 len=3670016
idx 10, ptn 7 name='misc' start=7569408 len=8192
idx 11, ptn 8 name='metadata' start=7577600 len=4096
idx 12, ptn 9 name='presistdata' start=7581696 len=2048
idx 13, ptn 10 name='vendor_a' start=7585792 len=524288
idx 14, ptn 11 name='vendor_b' start=8110080 len=524288
idx 15, ptn 12 name='userdata' start=8634368 len=18612224
idx 16, ptn 13 name='fbmisc' start=27246592 len=2048
idx 17, ptn 14 name='vbmeta_a' start=27250688 len=2048
idx 18, ptn 15 name='vbmeta_b' start=27254784 len=2048
`

by the way, the preceding log is output with i.MX8QuadXPlus MEK board with android_p9.0.0_2.0.0-ga release image. I f'ashed the image on windows without any problems encountered, and from the log it can be seen that, "bootloader0" is a partition that can be recognized by uboot running on the board.

B-Andras, 2020-09-15:
Thank you very much for the help and sorry for my misunderstanding.

I have copied the log of the debug UART during the flashing sequence

> Detect USB boot. Will enter fastboot mode!
> Detect USB boot. Will enter fastboot mode!
> switch to partitions #0, OK
> mmc0(part 0) is current device
> Detect USB boot. Will enter fastboot mode!
> switch to partitions #0, OK
> mmc0(part 0) is current device
> Detect USB boot. Will enter fastboot mode!
> 
> MMC erase: dev # 0, block # 8192, count 16 ... 
> 
> Caution! Your devices Erase group is 0x400
> The erase range would be change to 0x2000~0x23ff
> 
> 16 blocks erased: OK
> Detect USB boot. Will enter fastboot mode!
> Detect USB boot. Will enter fastboot mode!
> Starting download of 34304 bytes
> 
> downloading of 34304 bytes finished
> writing to partition 'gpt'
> Initializing 'gpt'
> switch to partitions #0, OK
> mmc0(part 0) is current device
> Writing 'gpt'
> 
> MMC write: dev # 0, block # 0, count 34 ... 34 blocks written: OK
> Writing 'gpt' DONE!
> flash backup gpt image successfully
> switch to partitions #0, OK
> mmc0(part 0) is current device
> Detect USB boot. Will enter fastboot mode!
> SATA isn't buildin
> Detect USB boot. Will enter fastboot mode!
> flash target is MMC:0
> Starting download of 1031168 bytes
> .......
> downloading of 1031168 bytes finished
> idx 0, ptn 0 name='gpt' start=0 len=2048
> idx 1, ptn 0 name='' start=0 len=0
> idx 2, ptn 0 name='all' start=0 len=61079552
> idx 3, ptn 0 name='bootloader' start=0 len=8192
> idx 4, ptn 1 name='dtbo_a' start=16384 len=8192
> idx 5, ptn 2 name='dtbo_b' start=24576 len=8192
> idx 6, ptn 3 name='boot_a' start=32768 len=98304
> idx 7, ptn 4 name='boot_b' start=131072 len=98304
> idx 8, ptn 5 name='system_a' start=229376 len=3670016
> idx 9, ptn 6 name='system_b' start=3899392 len=3670016
> idx 10, ptn 7 name='misc' start=7569408 len=8192
> idx 11, ptn 8 name='metadata' start=7577600 len=4096
> idx 12, ptn 9 name='presistdata' start=7581696 len=2048
> idx 13, ptn 10 name='vendor_a' start=7585792 len=524288
> idx 14, ptn 11 name='vendor_b' start=8110080 len=524288
> idx 15, ptn 12 name='userdata' start=8634368 len=18612224
> idx 16, ptn 13 name='fbmisc' start=27246592 len=2048
> idx 17, ptn 14 name='vbmeta_a' start=27250688 len=2048
> idx 18, ptn 15 name='vbmeta_b' start=27254784 len=2048
> =>

Thank you for sharing the log, the differences I found compared to mine is:
1 - the size of download (my log: 1031168 bytes, reference log: 1355776 bytes)
2 - in my case the isx3 partition it is called 'bootloader' and not 'bootloader0'

I hope this helps, but please let me know if you need any other information,

a09bdd, 2020-09-15:
yes, what matters is the difference between "bootloader" and "bootloader0".

I should briefly mention how uuu works here.
1. firstly, download "u-boot-imx8qxp-mek-uuu.imx" to the board RAM and have the board running with this image.
2. the board enters into uboot fastboot mode, uuu uses fastboot protocol to flash images to the corresponding partitions on emmc/sd, in which the "uboot-imx8qxp.imx" should be flashed to "bootloader0" partition.

below I will refer to some uboot code, the git repository tag is "p9.0.0_2.0.0-ga"

this "bootloader0" partition is hardcoded uboot image. the name is defined with the macro "FASTBOOT_PARTITION_BOOTLOADER", you can find it in:

${MY_ANDROID}/vendor/nxp-opensource/uboot-imx/include/fsl_fastboot.h

in this file, you can know that this macro can be defined as "bootloader" or "bootloader0", based on whether "CONFIG_ANDROID_AB_SUPPORT" is defined or not. this "CONFIG_ANDROID_AB_SUPPORT" is a macro indicate whether a/b slot feature is enabled. on imx8 based platforms running Android, "CONFIG_ANDROID_AB_SUPPORT" is defined. take 8qxp as an example, you can find it in:

${MY_ANDROID}/vendor/nxp-opensource/uboot-imx/include/configs/imx8qxp_mek_android.h

other information of the partition for uboot image can be found in below file:
${MY_ANDROID}/vendor/nxp-opensource/uboot-imx/drivers/usb/gadget/f_fastboot.c
` 
   strcpy(ptable[PTN_BOOTLOADER_INDEX].name, FASTBOOT_PARTITION_BOOTLOADER);

    ptable[PTN_BOOTLOADER_INDEX].start =
                bootloader_mmc_offset() / dev_desc->blksz;
    ptable[PTN_BOOTLOADER_INDEX].length =
                 ANDROID_BOOTLOADER_SIZE / dev_desc->blksz;
    ptable[PTN_BOOTLOADER_INDEX].partition_id = boot_partition;
    ptable[PTN_BOOTLOADER_INDEX].flags = FASTBOOT_PTENTRY_FLAGS_UNERASEABLE;
    strcpy(ptable[PTN_BOOTLOADER_INDEX].fstype, "raw");
`

in the "uuu_imx_android_flash.bat", I set the partitition name for uboot to be "bootloader0" for imx8 devices.
`
    if not %soc_name:imx8=% == %soc_name% (

        call :flash_partition bootloader0 || set /A error_level=1 && goto :exit
    ) else (
        call :flash_partition bootloader || set /A error_level=1 && goto :exit
    )
`

I tried the release release, no such issue is encountered, the partition for uboot is named "bootloader0", but in your condition, it "bootloader". so:
1. are the images you used our release images? especially the "u-boot-imx8qxp-mek-uuu.imx".
2. if you just want to work around this flash issue, I think you can get enough info from preceding comment.

nxpfrankli, 2021-02-11:
reopen if still problem.

closed 2021-02-11 by nxpfrankli