nxp-imx/mfgtools #223: UUU Frequent Failure after MFG kernel load.

ajithpv, 2020-09-15:
Hi,

We encounter with a strange issue as listed below:
1. When we flashed around 60 boards, 30 boards got failed with USB failure.
1. Almost all boards failed at below message:

```
2:2>Start Cmd:FB: ucmd setenv fastboot_dev mmc
2:2>Okay (0.004s)
2:2>Start Cmd:FB: ucmd setenv mmcdev 0
2:2>Okay (0.139s)
2:2>Start Cmd:FB: ucmd mmc dev 0
2:2>Okay (0.142s)
2:2>Start Cmd:FB: ucmd setenv loadaddr_os_cntr 0x86000000
2:2>Okay (0.004s)
2:2>Start Cmd:FB: ucmd setenv fastboot_buffer ${loadaddr_os_cntr}
2:2>Okay (0.003s)
2:2>Start Cmd:FB: download -f os/os_cntr_mfg_signed.bin
2:2>Okay (0.726s)
2:2>Start Cmd:FB: ucmd auth_cntr ${loadaddr_os_cntr}
2:2>Okay (0.109s)
2:2>Start Cmd:FB: ucmd setenv fastboot_buffer ${initrd_addr}
2:2>Okay (0.004s)
2:2>Start Cmd:FB: acmd ${kboot} ${loadaddr} ${initrd_addr} ${fdt_addr}
2:2>Okay (0.003s)
New USB Device Attached at 2:2
2:2>Fail Failure open usb device(1.448e+16s)
[?25h
```
We would like to know why the USB failure is happening at the end of MFG kernel an device tree (in almost all cases). Also, how to overcome this issue from either hardware/software side?

ajithpv, 2020-09-24:
@nxpfrankli - I would like to get some insight here.

nxpfrankli, 2020-09-24:
look like linux kernel USB driver have start correct.   Which platform do you use? 

commit bb8aaeb2ba407542f6cadeb2274927dd4280c0cd
Author: Ye Li <ye.li@nxp.com>
Date:   Thu May 21 03:26:34 2020 -0700

    MLK-24079 imx8mp: Disable DWC3 gadget before UUU jumping to kernel

    When UUU runs acmd command to boot into kernel, the fastboot never
    get chance to exit, so the usb gadget is not disconnected. This causes
    kernel USB gadget failed to work.

    Re-use iMX8MQ implementation to disconnect DWC3 gadget before jumping
    to kernel.

    Signed-off-by: Ye Li <ye.li@nxp.com>
    Reviewed-by: Jun Li <jun.li@nxp.com>

ajithpv, 2020-09-25:
Thank you @nxpfrankli for the response.

I'm using i.MXQXP based platform. Whether the above mentioned case is applicable to i.MXQXP?

Please note that, it is not always failing here...the error case is like 10 out of 36 boards where most of the failure happened in this part.

closed 2020-10-07 by ajithpv