nxp-imx/mfgtools #224: FBK and imx-uuu

ole00, 2020-09-16:
The PDF documentation mentions FBK protocol implemented in project imx-uuu. 

Here is the quote from the documentation:
"FBK: Android fastboot protocol, implement at initramfs. See project imx-uuu"

I can't find the project sources. Could you please publish a link to the source code/project of imx-uuu?

I assume the implementation ox imx-uuu is a software that resides in initramfs, and is able to execute remote commands sent over USB by another PC utilising uuu tool. Is that correct? 
Thank you.

nxpfrankli, 2020-09-16:
yes, https://github.com/NXPmicro/imx-uuc

nxpfrankli, 2020-09-16:
typo fixed and added link
https://github.com/NXPmicro/mfgtools/wiki/UUU-default-support-protocol-list

ole00, 2020-09-18:
Thank you for the links.

closed 2020-09-18 by ole00