nxp-imx/mfgtools #227: fastboot flash raw2sparse or -S parameter

StehleJochen, 2020-10-12:
Hi there,
we want to flash a image via fastboot to a eMMC partition in production.
till now we used the command: FB[-t 40000]: flash rootfs0 rootFS.img
But when the image is bigger than the ram it fails. If we use the -raw2sparse parameter it work again but slow because u-boot only copies 16M datablocks to ram and than write it to eMMC.

Is it possible to make these Datablocks bigger (e.g. 400M)?

nxpfrankli, 2020-10-12:
-raw2sparse convert to raw format to android sparse format. and each download 16M utile whole image. 

If your image already is sparse file, needn't -raw2sparse.
If your image is raw data image,  need -raw2sparse

StehleJochen, 2020-10-12:
ok thanks. Than i would use the raw2sparse.

Is it possible to use a bigger buffer than the 16M?

nxpfrankli, 2020-10-12:
Limited 16M is because avoid need long timeout value.  
I remember It is hardcode.

nxpfrankli, 2021-02-11:
if still have problem, please reopen.

closed 2021-02-11 by nxpfrankli