nxp-imx/mfgtools #228: No space left on memory

henrikoch, 2020-10-14:
Hello,

I am trying to flash a rootfs using uuu. My partition size is only 80 Mb and I have 256 Mb of RAM

I tried:

```
FBK: ucmd mkdir -p /mnt/rootfs1
FBK: acmd rootfspart=`cat /tmp/rootfspart`;node=`cat /tmp/node`;mount ${node}p${rootfspart} /mnt/rootfs1
FBK: acmd export EXTRACT_UNSAFE_SYMLINKS=1; tar -xzvf - -C /mnt/rootfs1
FBK: ucp  core-image-minimal.tar.gz t:- 
FBK: Sync
FBK: ucmd umount /mnt/rootfs1
FBK: DONE
```

but I got each time a tar: write error: No space left on device write: Broken pipe

Do you know a way to circumvent this problem ? or I cannot use uuu for my flashing process ? 

Thanks.

nxpfrankli, 2020-10-14:
what size after extract core-image-minimal.tar.gz?  The small file occupty more disk space.

nxpfrankli, 2020-10-14:
FBK: acmd rootfspart=`cat /tmp/rootfspart`;node=`cat /tmp/node`;mount ${node}p${rootfspart} /mnt/rootfs1
you should use **ucmd** to run mount.  

it is more than your partition size! 200M vs 80M
out of memory cause system crash.

henrikoch, 2020-10-14:
Sorry for the confusion: 

The size of the file compressed is 80 Mb and uncompressed around 200 Mb while the partition is 1.5 Gb. 

Is the file first downloaded on the RAM then flash to the partition or it is a step by step with small chunk of data ?

nxpfrankli, 2020-10-14:
No,  it is use pipe command.  
It looks like mount failure.

henrikoch, 2020-10-21:
The problem got resolved after changing ucmd command in my code. Thanks.

closed 2020-10-21 by henrikoch