nxp-imx/mfgtools #231: Why not using brew symbolic link on macOS?

roddehugo, 2020-11-06:
Hi!

Thank you for providing the UUU tool for all major platforms including macOS.

While testing to build and use `uuu` from `master` branch, with a freshly `brew`ed openssl dependancy, I noticed the build failed because my local Cellar having `openssl@1.1/1.1.1h` instead of `openssl@1.1/1.1.1h` as written in your [`README.md`](https://github.com/NXPmicro/mfgtools/blob/master/README.md#macos)

Here is an example:

    ls -la /usr/local/opt/openssl
    lrwxr-xr-x  1 roddehugo  admin    28B Oct 27 12:25 /usr/local/opt/openssl -> ../Cellar/openssl@1.1/1.1.1h

Hence, I tried to build with:

    cmake -DOPENSSL_ROOT_DIR=$(brew --prefix)/opt/openssl . && make

Thankfully, it works as expected. 

IMO, using symbolic link from brew is more resilient again changes. Any reason not using `$(brew --prefix)/opt/openssl` instead of`$(brew --cellar)/openssl@1.1/1.1.1g` in your docs (and Travis CI)?

I'd be happy to make a PR if you're in favor of this change.

All the best,
Hugo

roddehugo, 2020-11-06:
Attached a log to assert CMake actually finds OpenSSL with symlink:

```
[...]
-- The C compiler identification is AppleClang 12.0.0.12000032
-- The CXX compiler identification is AppleClang 12.0.0.12000032
[...]
-- Found OpenSSL: /usr/local/opt/openssl/lib/libcrypto.dylib (found version "1.1.1h")
```

nxpfrankli, 2020-11-06:
Please submit PR

PR #232 (2020-11-06): use macOS brewed openssl symbolic link to resolve root dir for cmake
By using `$(brew --prefix)/opt/openssl` symbolic link instead of hard
coded Cellar path to fix version, CMake will continue to find OpenSSL
even if the version changes.

Closes #231.
commit 3711e0b0 (2020-11-06): use macOS brewed openssl symbolic link to resolve root dir for cmake

By using `$(brew --prefix)/opt/openssl` symbolic link instead of hard
coded Cellar path to fix version, CMake will continue to find OpenSSL
even if the version changes.

Closes #231.
closed 2020-11-06 by nxpfrankli