nxp-imx/mfgtools #233: "Fail Failure open usb device" During New USB Device Attached

ajithpv, 2020-11-09:
Hi,

We are doing the parallel flashing (2 boards at the same time) on the PCB level. One panel/PCB contains 2 boards and both the boards will flash in parallel using UUU (with -m option to give specific USB port numbers).
Please note that, we are using batch file to call the UUU.exe flashing from the Windows PC. We have a windows PC software tool which will call these 2 batch files (each batch file dedicated to one i.MX8DX board) to run it parallel. In this way, we are achieving parallel flashing (i.e 2 boards will flash at the same time).

We have below case where the same UUU package is succeeding on one board and failing on another board (in the same panel/PCB) with USB attach failure! This is random and some times both the boards will pass without any issues.

**Failure case:**
_2:2>Start Cmd:FB: acmd ${kboot} ${loadaddr} ${initrd_addr} ${fdt_addr}
2:2>Okay (0.004s)
**New USB Device Attached at 2:2
New USB Device Attached at 2:2**
2:2>Fail Failure open usb device(1.845e+16s)
2:2>Start Cmd:FBK: ucp os/mksdcard.sh t:/tmp
DIR
13973%2:2>Okay (0.002s)
2:2>Start Cmd:FBK: ucmd chmod +x /tmp/mksdcard.sh
2:2>Okay (0.007s)
2:2>Start Cmd:FBK: ucmd /tmp/mksdcard.sh /dev/mmcblk0
/mksdcard.sh: line 1: ï»¿#!/bin/sh: No such file or directorrecords in
1+0 records out
 bytes (1.0 kB, 1.0 KiB) copied, 0.00553363 s, 185 kB/s[?25h_

**Pass case:**
_2:1>Start Cmd:FB: acmd ${kboot} ${loadaddr} ${initrd_addr} ${fdt_addr}
2:1>Okay (0.004s)
New USB Device Attached at 2:1
2:1>Start Cmd:FBK: ucp os/mksdcard.sh t:/tmp
DIR
13973%2:1>Okay (0.001s)
2:1>Start Cmd:FBK: ucmd chmod +x /tmp/mksdcard.sh
2:1>Okay (0.008s)
2:1>Start Cmd:FBK: ucmd /tmp/mksdcard.sh /dev/mmcblk0
/mksdcard.sh: line 1: ﻿#!/bin/sh: No such file or directorrecords in
1+0 records out
 bytes (1.0 kB, 1.0 KiB) copied, 0.005373 s, 191 kB/sChecking that no-one is using this disk right now ... OK
Disk /dev/mmcblk0: 3.6 GiB, 3833593856 bytes, 7487488 sectors_

May I know why the failure case is having 2 times the "New USB Device Attached at 2:2" where as success case only have it once?

nxpfrankli, 2020-11-09:
I think linux kernel side cause lost USB connection. Does one boards stable enough?

ajithpv, 2020-11-09:
Thank you @nxpfrankli for the reply.

This issue is random and seen mainly with parallel flashing where 1 board will pass and other will fail. Please note that both boards are flashing with the same software.

Could you please let me know what that 2 times "New USB Device Attached at 2:2" message means?  Ideally it should be only one correct?

nxpfrankli, 2020-11-09:
It means usb disconnect and reconnected again.  It should be happen once.

ajithpv, 2020-11-09:
@nxpfrankli Thank you. We are seeing this message twice just before the failure in all failure cases. For success, this message comes just once and start the operation.

Whether this provide any insight or clue on why it is happening? Also, could you please let me know how to debug this strange issue? I'm totally clueless on this random error.

ajithpv, 2020-11-10:
@nxpfrankli - I have added -V to the uuu.exe while flashing. During flashing, we recieved the below error:

`libusb: error [windows_transfer_callback] detected I/O error 1167: [1167] The device is not connected.`

Could you please let me know what this error message means?

Also, whether any specific USB package or driver is required on Windows 10 to resolve this random libusb issue?

nxpfrankli, 2020-11-10:
It should be disconnect by device. Not related host side.

ajithpv, 2020-11-10:
Thank you @nxpfrankli for your support.

Do you suspect anything on the device side for this issue? Whether you face this kind of issue during your testing or flashing?

nxpfrankli, 2020-11-10:
Yes,  long time ago meet some similar problem. but it should be fixed at kernel USB driver. Simplest method, you try a board first.

ajithpv, 2020-11-12:
Thank you @nxpfrankli for your support.

**Quick update** - The above I/O error (_libusb: error [windows_transfer_callback] detected I/O error 1167: [1167] The device is not connected_.) is coming even for the success flashing during the initial time. I believe this is due to the _FB_ (MFG bootloader) to _FBK_ (MFG Kernel) transfer where the USB is intentionally disconnecting after FB and connecting back at FBK starting.

Could you please confirm my understanding is correct and is expected at the beginning?

ajithpv, 2020-11-12:
@nxpfrankli - I would like to isolate the root cause for the "_Fail Failure open usb device_" issue. Hence, please let me know whether any of the below case (using for parallel flashing in our case) will cause this issue:

1. I'm using 2 batch files and each will execute UUU.exe (version 1.3.102) with specific USB ports (with -m option). Both batch files are running as different process and in parallel.

2. We are flashing at the panel level and one panel is having 2 boards. The panel is not having the USB connectors. We are placing the USB wires on the panel test points for the flashing (because USB connectors are not part of the final board).

3. The UUU.exe executing from Windows 10 PC. However, this PC is not connected to any Network due to the security restriction. Hence, Windows 10 will not get any service pack or updates.

Could you please let me know any of this will lead to the in-between USB issue that we are observing (_Fail Failure open usb device_)? This issue is happening randomly but failing always after the MFG kernel loads (during the partition stage as mentioned in above thread).

nxpfrankli, 2020-11-12:
kernel boot need disconnect USB and connect again.  I guest this flow cause twice reset.  I suggest just use one board to do test to isolate the problem.

ajithpv, 2020-11-13:
@nxpfrankli - Thank you for your suggestion.
Please note that, the panel always comes with 2 boards on single PCB. After flashing, it will break and make it as 2 boards.

We have flashed the single board (board with enclosure and USB connector) around 12 times continuously (one after another) with the same flashing package. All flashing got successfully completed.

The USB issue that observing (i.e. _Fail Failure open usb device_) is coming with parallel flashing.

closed 2021-05-04 by ajithpv