nxp-imx/mfgtools #234: timeout for FB: works from shell mode but not from command line

hpalacio, 2020-11-11:
When using **uuu** in shell mode, I can pass a timeout (in ms) to the FB: command as in 
```
~$ uuu -s
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.43-0-ga9c099a

U>FB[-t 5000]: ucmd <uboot_commmad>
```
in the example, it will wait up to 5 seconds for <uboot_command> command to complete.

However if I try to run the same on a full command, it fails:
```
~$ uuu FB[-t 5000]: ucmd <uboot_command>
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.43-0-ga9c099a

Wait for Known USB Device Appear...
Error: fail open file: >FB[-t
```

Am I missing something? Notice this is the timeout for the FB: command, not the timeout of **uuu** waiting for a USB device to attach.

nxpfrankli, 2020-11-11:
"FB[-t 5000]:"

hpalacio, 2020-11-11:
Thanks, that works!