nxp-imx/mfgtools #235: UUU not working on closed i.MX 6UL EVK

danie007, 2020-11-12:
Hi,
I'm experiencing issues with `UUU` tool as it's not working (it waits indefinitely) on closed [i.MX 6UL EVK](https://www.nxp.com/design/development-boards/i-mx-evaluation-and-development-boards/i-mx6ultralite-evaluation-kit:MCIMX6UL-EVK).
[image]
The script we're using is given below:
```shell
uuu_version 1.2.39

SDP: boot -f uboot_signed.imx

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev 1
FB: ucmd mmc dev 1
FB: flash bootloader uboot_signed.imx
FB: Done
```
The same script is working fine (flashed bootloader in SD Card) in a open, secure board and `uboot_signed.imx` is also manually verified to be working.

The `fastboot` parameters are as follows:
```shell
CONFIG_FASTBOOT_BUF_ADDR=0x83800000
CONFIG_FASTBOOT_BUF_SIZE=0x40000000
```

Detailed log is given below to aid in your diagnostics:
```shell
$uuu -v ./signed_write_boot_sd.lst 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.43-0-ga9c099a

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
Wait for Known USB Device Appear...
New USB Device Attached at 1:124
1:124>Start Cmd:SDP: boot -f uboot_signed.imx
6400%1:124>Okay (2.276s)
```

_Originally posted by @danie007 in https://github.com/NXPmicro/mfgtools/issues/153#issuecomment-725909075_

nxpfrankli, 2020-11-16:
Let's check with ROM team

nxpfrankli, 2020-11-17:
```
int SDPCmdBase::get_status(HIDReport *p, uint32_t &status, uint8_t report_id)
{
	m_input.resize(1025);
	m_input[0] = report_id;
	int ret = p->read(m_input);
	if (ret < 0)
		return -1;

	if (m_input.size() < (1 + sizeof(uint32_t)))
	{
		set_last_err_string("HID report size is too small");
		return -1;
	}

	status = *(uint32_t*)(m_input.data() + 1);

printf("0x%x \n", status); //can you add this line, run with uuu -v, send back log 
	return 0;
}

```

danie007, 2020-11-17:
Hi @nxpfrankli 
> printf("0x%x \n", status); //can you add this line, run with uuu -v, send back log

I've added this line to `int SDPCmdBase::get_status(HIDReport *p, uint32_t &status, uint8_t report_id)` function in `sdp.cpp` and the log is as follows:
```shell
$./uuu -v ./signed_write_boot_sd.lst 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.43-9-gb1d7956

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
Wait for Known USB Device Appear...
New USB Device Attached at 1:124
1:124>Start Cmd:SDP: boot -f uboot_signed.imx
209%0x12343412
0x128a8a12
100%0x12343412
0x88888888
6400%0x12343412
0x900dd009
6400%0x12343412
0xa00c33
1:124>Okay (2.279s)
^C
# Waits indefinitely
```

CC @tkmozhi

danie007, 2020-11-20:
@nxpfrankli any updates on this?

mike-scott, 2020-12-17:
@nxpfrankli @danie007 I'm working on i.MX6ULL/ULZ using a custom .config with split SPL / u-boot.bin.

To boot the signed SPL on closed HW via `uuu` we needed to add the DCD signing block as well as load SPL like this in the .uuu script:
```
SDP: boot -f SPL -dcdaddr 0x0907000 -cleardcd
```

Perhaps, this can point you in the right direction.
(Possibly the same issue as: https://github.com/NXPmicro/mfgtools/issues/242)

nxpfrankli, 2020-12-17:
need both change dcdaddr and cleardcd?

Maybe only cleardcd work?

mike-scott, 2020-12-17:
> need both change dcdaddr and cleardcd?
> 
> Maybe only cleardcd work?

When I remove the dcdaddr, I see the following HAB event:
```
--------- HAB Event 1 -----------------
event data:
        0xdb 0x00 0x1c 0x42 0x33 0x18 0xc0 0x00
        0xca 0x00 0x14 0x00 0x02 0xc5 0xff 0x00
        0x00 0x00 0x0c 0xa0 0x00 0x90 0x70 0x00
        0x00 0x00 0x01 0xe8

STS = HAB_FAILURE (0x33)
RSN = HAB_INV_SIGNATURE (0x18)
CTX = HAB_CTX_COMMAND (0xC0)
ENG = HAB_ENG_ANY (0x00)
```

nxpfrankli, 2020-12-18:
How about use DCDaddress and remove cleardcd option?

mike-scott, 2020-12-18:
> How about use DCDaddress and remove cleardcd option?

Also results in HAB events (security error)

jmonkman, 2021-01-19:
I just ran into this issue on mx7d. I had a closed device and needed to run UUU to flash u-boot.

It looks to me that there are 2 issues in the uuu source code.
1: The way the -cleardcd flag is handled looks completely wrong.
2: The DCD pointer in the IVT was not being cleared when it should be

I've attached a patch that fixes it for me. I'm not sure that the patch is correct, because I'm not 100% clear on what uuu is supposed to be doing.

With this patch, I am able to recover my board. I did have to create a special signed image, as the DCD is handled differently when booting with SDP compared to booting from flash. In a nutshell, uuu writes the DCD to OCRAM and runs it from there, then loads the image, clears the DCD pointer in the image's IVT, and jumps to the authenticate image in ROM. To make this work, I had do the following:
     - create a copy of my uboot image and set the DCD pointer to 0x00000000. That is the 4th word in the IVT
     - break up the "blocks" line in my CSF file for the code signing tool. Here is mine. This will have to be modified for a different CPU and image:
    Blocks = 0x877ff400 0x000 0x02c "/tmp/files/u-boot-nodcdptr.imx", \    # includes IVT and boot data
    	     0x00911000 0x02c 0x1c4 "/tmp/files/u-boot-nodcdptr.imx", \       # use OCRAM address for DCD
             0x87800000 0xc00 0x100 "/tmp/files/u-boot-nodcdptr.imx"          # remainder of image - must include entry point
    - Once I get the output from CST, I append it to the UNMODIFIED u-boot image. uuu needs the unmodified IVT so that it knows where to get the DCD, but uuu will clear that pointer after it loads the image. That's why the signature has to be based on an image with the DCD cleared.
    - When I boot the image, use the '-cleardcd' flag:
        SDP: boot -f /tmp/files/u-boot-signed-uuu.imx -cleardcd

[patch-cleardcd.diff.txt](https://github.com/NXPmicro/mfgtools/files/5833211/patch-cleardcd.diff.txt)

nxpfrankli, 2021-01-19:
I don't understand well. Can you send pull request and clean up unnecessary change. 

I think original logic should be correct.  ROM skip DCD should work at i.MX7D.

jmonkman, 2021-01-19:
I was mistaken. I edited my comment above. Disregard my previous post.

jynik, 2021-02-24:
@mike-scott - Could you elaborate on how you determined the correct `-dcdaddr` value?  

Looking at [Figure 6 in AN4581](https://www.nxp.com/docs/en/application-note/AN4581.pdf#page=9), it seems clear that I need to somehow help the device locate the DCD table in my signed image...  should this be the value shown in the `DCD Blocks` line from a U-Boot imx image log (below) ?

~~~
Image Type:   Freescale IMX Boot Image
Image Ver:    2 (i.MX53/6/7 compatible)
Mode:         DCD
Data Size:    356448 Bytes = 348.09 KiB = 0.34 MiB
Load Address: 877ff420
Entry Point:  87800000
HAB Blocks:   0x877ff400 0x00000000 0x00054c00
DCD Blocks:   0x00910000 0x0000002c 0x000001e8
~~~

I believe I'm seeing a similar issue to that discussed here with an i.MX6ULL platform, using UUU @ db1953a5def7494a8254c3f07ee25c7836ae3e2a.

My chip is fused with SRK values, but is not closed yet. I am looking to have confidence that I can boot a signed image via both NAND and SDP before closing the device.

When booting my `u-boot-dtb.imx.signed` image (single Proper image, no SPL) from NAND, I see no HAB events reported by `hab_status`. 
However, when booting via SDP I see a single event entry.

~~~
$ uuu -v -V u-boot-dtb.imx.signed
~~~

~~~
IMX6ULL-SOM U-Boot > hab_status

Secure boot disabled

HAB Configuration: 0xf0, HAB State: 0x66

--------- HAB Event 1 -----------------
event data:
        0xdb 0x00 0x14 0x42 0x33 0x0c 0xa0 0x00
        0x00 0x00 0x00 0x00 0x00 0x91 0x00 0x00
        0x00 0x00 0x01 0xe8

STS = HAB_FAILURE (0x33)
RSN = HAB_INV_ASSERTION (0x0C)
CTX = HAB_CTX_ASSERT (0xA0)
ENG = HAB_ENG_ANY (0x00)
~~~

mike-scott, 2021-02-25:
@jynik Can you try altering your CSF file that you're using to sign with cst to (under `[Header]`):
```
Engine = SW
```

i.MX6UL[L/Z] SoC doesn't have a CAAM IP block and needs to use this setting.

jynik, 2021-02-25:
@mike-scott Huh. That is a very good point.

I had been using the [CSF example from the U-Boot](https://github.com/jynik/u-boot/blob/master/doc/imx/habv4/csf_examples/mx6_mx7/csf_uboot.txt#L7) which sets it to CAAM.

Will try and report back.

That being said -- how on earth am I successfully booting from NAND then (while booting via UUU fails) ? 😖

jynik, 2021-02-26:
@mike-scott Tried making the CAAM -> SW change in the CSF, but still no luck with UUU.  I need to delve deeper into understanding why `-dcdaddr` and `-cleardcd` are needed... still feel like I'm doing something incorrect here on the UUU side of things.

jynik, 2021-03-16:
@jmonkman - With the UUU code as-provided, do you still need to modify the CSF and sign an image with the DCD pointer in the IVT zeroized? 

What you describe doing sounds similar in Appendix E: Freescale Manufacturing Tool of a 10/2015 version of AN4581, which was removed in later revisions...

@nxpfrankli  - Any chance you could take a look at my preceding posts and comment further?

danie007, 2021-12-15:
@nxpfrankli 

### Any update here?

We're facing this issue again on the **i.MX 6SX Sabre-SD EVK**.

[image]

---

CC @tkmozhi

jynik, 2021-12-19:
@danie007 - Does my previous post apply at all?  

If not, would you mind sharing a some screenshots (redacted if as needed) of a Wireshark usbmon capture showing the summary of the SDP HID transfers? This probably won't show much, beyond confirming that the timeout occurs upon sending the `jump` operation that'll kick off what probably ends up being an auth failure?

danie007, 2021-12-19:
@jynik  thanks for the consideration, I’ll check and update

danie007, 2021-12-22:
# Fixed with CSF tweak 🌈

@jynik thank you for your providing useful insights ✌🏾. However, I had to make some minor changes to get it boot on a closed `i.MX 6SX SABRE-SD`.

## My sequence:

1. Cleared DCD pointer on `u-boot.imx`

```bash
dd if=/dev/zero of=u-boot.imx.cleardcd seek=12 bs=1 count=4 conv=notrunc
```

2. Appended separate authentication block for DCD (from OCRAM)

```bash
Blocks = 0x877FF400 0x00 0x97C00 "u-boot.imx.cleardcd", \
#        Separate DCD block authentication
         0x910000 0x2C 0x220 "u-boot.imx.cleardcd"
```

3.  Concatenated the original `u-boot.imx` and the CSF binary
4.  Boot with `-cleardcd` flag on UUU tool

@nxpfrankli you could have point us to the **HABv4 closed chip support** on [UUU.pdf](https://github.com/NXPmicro/mfgtools/releases/tag/uuu_1.4.165) that contain the useful information. See the excerpt below 👇🏽:
[image]

> So... it's not really a UUU bug/issue insomuch as added steps apparently being necessary?

I too feel this now 😬

---

CC @tkmozhi

closed 2021-12-22 by danie007