nxp-imx/mfgtools #236: Script must be in current directory

jmonkman, 2020-11-13:
If I want to use my own command script and run it like 'uuu script.uu', it works fine.
If I do 'uuu scripts/script.uu', it fails because it adds 'scripts/' to every pathname in the script.

For example, my script includes this line:
> boot -f /projects/yocto/build/tmp/deploy/images/sabre/u-boot.imx

When I run 'uuu scripts/script.uu', I get this message:
> Error: fail open file: >scripts//projects/yocto/build/tmp/deploy/images/sabre/u-boot.imx

The location of the script should not cause any changes to the paths in the script. That could lead to very confusing behaviour if ended up as a valid, but wrong filename

nxpfrankli, 2020-11-14:
all file support related to uuu script. If you want to use absolute path, please add > before path

nxpfrankli, 2021-02-11:
if still have problem, please reopen.

closed 2021-02-11 by nxpfrankli