nxp-imx/mfgtools #237: UUU tool only working when the files in same directory

danie007, 2020-11-18:
Hi,

When I try to run the UUU tool with `lst` file pointing to different location for `u-boot` and `image`.
I've given full paths for the files:
 ```
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.43-9-gb1d7956

Wait for Known USB Device Appear...
Error: fail open file: >/home/ux/Documents/cst_test//home/ux/Documents/cst_test/cst_encrypt_sign/tmp/u-boot.imx
```

As you can see `/home/ux/Documents/cst_test//home/ux/Documents/cst_test/cst_encrypt_sign/tmp/u-boot.imx` here `UUU` tool appends the current directory to the file location.

**NOTE**: It's working fine if all the files are in same `dir` or when we give the relative paths. I'm using script to automate `UUU` so it'll be easier if I can give the full paths.

danie007, 2020-11-18:
TODO: Try this suggested solution: https://github.com/NXPmicro/mfgtools/issues/236#issuecomment-727139060

nxpfrankli, 2020-11-18:
Can you add ">" before your full path in lst.  >/home/ux/Documents/cst_test//home/ux/Documents/cst_test/cst_encrypt_sign/tmp/u-boot.imx

danie007, 2020-11-20:
@nxpfrankli Thanks for the solution. It's working now.

closed 2020-11-20 by danie007