nxp-imx/mfgtools #238: How to add delay for FBK command?

ajithpv, 2020-11-19:
Hi,

I would like to know, how we can add delay for the FBK command.

So far, I understood how to add delay for initial timeout (using -t) and FB delay (using -t with FB command).

However, I didn't find how to add delay for FBK command.

nxpfrankli, 2020-11-19:
Do you means timeout? 

FBK needn't timeout because linux can return busy status every second.  uboot is single thread and can't return busy status when running a command.

ajithpv, 2020-11-19:
Thank you @nxpfrankli for your response.

Does this means timeout is not required for FBK? 

Also, what is the default timeout time for FBK and FB (if we didn't specify the -t with FB)?

nxpfrankli, 2020-11-19:
FBK and FB default value is 2s.

closed 2020-11-20 by ajithpv