nxp-imx/mfgtools #240: HID(W):LIBUSB_ERROR_TIMEOUT on IMX8MNANO

mickeykkim, 2020-11-30:
I'm having difficulty flashing the example linux images to the IMX8MNano with an SD card. 

I am using vagrant with ubuntu 20.04. The switch on board is set to 1000 (SDP).

Is there something I'm missing? The following is my log:

```
vagrant@ubuntu-focal:~/uuu$ sudo ./uuu -v -b spl L5.4.47-2.2.0_images_MX8MNEVK.zip/uuu.auto-imx8mnevk
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.43-0-ga9c099a

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001

Run built-in script:

uuu_version 1.2.39

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f L5.4.47-2.2.0_images_MX8MNEVK.zip/uuu.auto-imx8mnevk

# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM
SDPS: boot -f L5.4.47-2.2.0_images_MX8MNEVK.zip/uuu.auto-imx8mnevk

# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
# {
SDPU: delay 1000
SDPU: write -f L5.4.47-2.2.0_images_MX8MNEVK.zip/uuu.auto-imx8mnevk -offset 0x57c00
SDPU: jump
SDPU: done
# }

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f L5.4.47-2.2.0_images_MX8MNEVK.zip/uuu.auto-imx8mnevk -skipspl
SDPV: jump
SDPV: done
# }

Wait for Known USB Device Appear...
Decompress file:uuu.auto-imx8mnevk
100%New USB Device Attached at 1:1
1:1>Start Cmd:SDPS: boot -f L5.4.47-2.2.0_images_MX8MNEVK.zip/uuu.auto-imx8mnevk
1:1>Fail HID(W):LIBUSB_ERROR_TIMEOUT(1.005s)
```

nxpfrankli, 2020-11-30:
sudo ./uuu -v L5.4.47-2.2.0_images_MX8MNEVK.zip/uuu.auto-imx8mnddr4evk

concise77, 2021-01-12:
Hello，Do you have the answer?

nxpfrankli, 2021-01-12:
what's answer?

closed 2021-07-28 by nxpfrankli