nxp-imx/mfgtools #241: reboot uboot twice

JanneTerho1, 2020-11-30:
Hi,

Is there any way to reboot device and boot it end of the script with new uboot file and make uboot commands again?

After uuu script erase and flash nand I need to reboot device and start uboot again. I need to make saveenv in uboot to save uboot environments to flash.

Is there any way to start uboot again in the end of the script? I tried it but every SDP commands runs before FB commands and every FB commands runs before FBK commands. Even they are end of the script. 
So after first SDP command uuu-tool will run last SDP command immediately and goes error.

nxpfrankli, 2020-11-30:
Not support.  You can use fb: acmd: reboot to reboot system. but can't use difference uboot file.  

if only want to saveevn to flash, you can use https://bootlin.com/blog/mkenvimage-uboot-binary-env-generator/ to generate environment variable binary. Then use uuu to write it to flash.

JanneTerho1, 2020-12-01:
Thanks for fast reply.

It could be okay if it will reboot same uboot. I tried fb: acmd reboot, but it reboots before initramfs kernel starts, even fb: acmd reboot command is end of the script file.

Maybe I use CONFIG_USE_DEFAULT_ENV_FILE in uboot and use same file with mkenvimage in uuu flash script after nand partition erase. I don't want use env file in the running kernel, because we have two partition for updating device and we need sometimes change uboot environments with fw_setenv. Same environments in many places not sound good.

nxpfrankli, 2020-12-01:
Which platform do you use?  Some chip have some persistent register in SRTC.  You can write some information into it.  after reboot, your uboot read it back and choose difference environment variable.

JanneTerho1, 2020-12-01:
My chip is nxp imx6ull.

Nice to know about SRTC. I will check it also.

Thanks for mkenvimage info. Now it works. I use BR2_TARGET_UBOOT_ENVIMAGE from buildroot, so I get uboot-env.bin from buildroot. Then I just write it to nand in script 
FBK: ucmd nandwrite -p /dev/mtd2 /tmp/uboot-env.bin

danie007, 2020-12-02:
> fb: acmd: reboot to reboot system

@nxpfrankli I added the above line in my `UUU` script and ran it in **i.mx 8MM Mini EVK**. I got the following error:

```shell
Error: Unknown Command:FB: acmd: reboot

Error: Unknown Command:FB: ucmd: reboot
```

Could you kindly tell us how to reset the i.MX 8M with UUU tool.

JanneTerho1, 2020-12-02:
I think it's
`FB: acmd reboot`
not
`FB: acmd: reboot`

danie007, 2020-12-02:
Hi @JanneTerho1 

_Thanks for the quick reply._

> FB: acmd reboot

`reboot` doesn't seem to work in `i.MX 8MM` I got this: `Unknown command 'reboot' - try 'help'`. However the syntax is proven right as I put reset instead of reboot (i.e. `FB: acmd reset`) everything working as expected now.

BTW, would you mind explaining about `acmd` a bit?

nxpfrankli, 2020-12-02:
acmd return success when  uboot received cmd. 
ucmd return success after cmd run. 

because reset allow system reboot. So use acmd

danie007, 2020-12-03:
@nxpfrankli thanks for the clarification.

JanneTerho1, 2020-12-03:
Envimage was good in kernel side, but I need to change in uboot also. If I change nand partition sizes, I need to change it to uboot config and uboot-env.txt. Also if I use env default -a in uboot, it will return uboots environments, not the generated environments from file.

I tried to use uboot CONFIG_DEFAULT_ENV_FILE, but it didn't work from same uboot-env.txt.

I think that easiest way could be to made it in uuu-script, but I think it's not possible.

JanneTerho1, 2020-12-03:
I can't get reboot working.

FB: acmd reset and FBK: acmd reboot will show that commads executes correctly on device, but nothing happend. In normal running device I can boot device with those commands, but not in initramfs via uuu-tool.

nxpfrankli, 2021-02-11:
if still have problem, please reopen.

closed 2021-02-11 by nxpfrankli