nxp-imx/mfgtools #242: UUU tool working in i.MX 8MM but not in i.MX 6UL

danie007, 2020-12-14:
Hi,

We've tried flashing i.MX 6UL with `UUU` tool and it's not working on closed i.MX 6UL (ref: #235 ). However when we try to work on i.MX 8MMini EVK from NXP, UUU tool works seamlessly on closed device. So, we would like to know the reason and workaround for making UUU work with closed i.MX 6UL.

Feel free to revert for any details. Thanks in advance.

danie007, 2020-12-14:
@nxpfrankli do you have any idea on this?

nxpfrankli, 2020-12-14:
ping rom team again, wait for feedback.

nxpfrankli, 2020-12-15:
after failure, please use below command to dump log buffer

uuu sdp: RDMEM  -addr 0x00901D14 --format 32

uuu sdp: RDMEM  -addr 0x00901D18 --format 32

...

total 64 words.  (0x00901E14)

danie007, 2020-12-15:
Hi @nxpfrankli ,

> after failure

Thanks for the quick update. I ran the commands you mentioned above (I had to change `--format` to `-format` tho) after I kill my script (about 1 min) as my script tends to hang indefinitely.

RESULTS:
```shell
uuu sdp: RDMEM -addr 0x00901D14 -format 32 -> Value of address 0x00901D14: 0x00010001

uuu sdp: RDMEM -addr 0x00901D18 -format 32 -> Value of address 0x00901D18: 0x000200CC
```

The below screenshot is captured after running the above commands.
[image]

Feel free to revert for more details.

CC @tkmozhi

nxpfrankli, 2020-12-16:
please dump all memory from 0x00901D18 -- 0x00901E14

danie007, 2020-12-16:
> dump all memory

Would you mind elaborating the steps?

danie007, 2021-12-15:
@nxpfrankli 

### Any update here?

We're facing this issue again on the **i.MX 6SX Sabre-SD EVK**.

[image]

---

CC @tkmozhi

nxpfrankli, 2021-12-15:
One colleague is working on 7d with similar problem. 
Still not figure out the root cause.

danie007, 2021-12-22:
@nxpfrankli FYI: https://github.com/NXPmicro/mfgtools/issues/235#issuecomment-999315304