nxp-imx/mfgtools #244: CMake 3.10 (Ubuntu 18.04 LTS) vs add_compile_definitions()

ALTracer, 2020-12-16:
This statement is introduced in CMake 3.12, while Ubuntu 18.04 LTS (bionic) still has only CMake 3.10 in its repositories.
https://github.com/NXPmicro/mfgtools/blob/63b1d3cf473369d13a778fc32d46ee5d398128b4/uuu/CMakeLists.txt#L76
Something along these lines allows to compile from source on bionic.
```patch
-add_compile_definitions(TARGET_PATH="${CMAKE_INSTALL_PREFIX}/bin/uuu")
+if(CMAKE_VERSION VERSION_GREATER_EQUAL 3.12.0)
+    add_compile_definitions(TARGET_PATH="${CMAKE_INSTALL_PREFIX}/bin/uuu")    
+else()
+    add_definitions(-DTARGET_PATH="${CMAKE_INSTALL_PREFIX}/bin/uuu")
+endif()
```
Now we could even lower the `cmake_minimum_required` to `VERSION 3.10`, unless CMake 3.12 treats your C++14 sources significantly differently and is absolutely required. 

Originally encountered errors for SEO:
`CMake 3.12 or higher is required.  You are running version 3.10.2`
`Unknown CMake command "add_compile_definitions".`

ALTracer, 2020-12-16:
Apparently CMake version requirement has been bumped recently (30Oct) from 3.4 straight to 3.12 in 0a57288edd20b2f90e4a4d209d0104782a299302
The only other CMakeLists-related change in that commit is `CMAKE_SKIP_RPATH` which is supported even since CMake 3.0 or so.

I see no reason to depend on 3.12 for this project even though latest release is 3.19.1. You may argue otherwise.
For everyone else buliding on Ubuntu bionic, feel free to test for yourselves. I am not an expert in CMake or C++.  We don't use mfgtools/uuu to flash our targets (yet).

nxpfrankli, 2020-12-16:
I am no idea. Can check with origin patch owner?