nxp-imx/mfgtools #245: UUU built on Windows 10 not working outside build dir

danie007, 2020-12-21:
Hi @nxpfrankli 

I've tried building `uuu` on Windows and this gives me the following error whenever, I try running outside the build directory. 
[image]

It would be helpful if you could point me out how to attach the `libusb.dll`(*it present in build folder*) into the `uuu.exe`

Thanks in advance.

nxpfrankli, 2020-12-23:
Please use uuu_static.sln

danie007, 2020-12-24:
Hi @nxpfrankli 

Thanks for pointing out our mistake, after building `uuu` with [uuu-static-link.sln](https://github.com/NXPmicro/mfgtools/blob/master/msvc/uuu-static-link.sln) everything work as intended.

CC @tkmozhi

closed 2020-12-24 by danie007