nxp-imx/mfgtools #246: Unable to flash with MacOS Big Sur

almatthew, 2020-12-27:
I'm unable to flash an i.mx6 device using `uuu` version `uuu_1.4.43` in MacOS Big Sur.  Specifically, I get the error:

```
20:34    1/ 2 [Failure claim interface               ] ... 
```

Digging a bit more in `dmesg`, I find:

```
[  977.389928]: 000977.389927 uuu@: AppleUSBHostUserClient::openGated: could not open provider Freescale Flash. provider already opened for exclusive access by a kernel client
```

Looking at `ioreg`, I find:

```
| +-o AppleUSB20HubPort@14340000  <class IORegistryEntry:IOService:AppleUSBHostPort:AppleUSBHubPort:AppleUSB20HubPort, id 0x100000ad2, registered, matched, active, busy 0 (31 m$
  |   +-o SE Blank ARIK@14340000  <class IORegistryEntry:IOService:IOUSBNub:IOUSBDevice, id 0x100000b1a, registered, matched, active, busy 0 (31 ms), retain 32>
  |     | {
  |     |   "sessionID" = 44121558910
  |     |   "USBSpeed" = 3
  |     |   "IOServiceLegacyMatchingRegistryID" = 4294970140
  |     |   "idProduct" = 84
  |     |   "iManufacturer" = 1
  |     |   "bDeviceClass" = 0
  |     |   "IOPowerManagement" = {"PowerOverrideOn"=Yes,"CapabilityFlags"=32768,"MaxPowerState"=2,"DevicePowerState"=2,"DriverPowerState"=0,"ChildrenPowerState"=1,"CurrentPowerS$
  |     |   "bcdDevice" = 1
  |     |   "bMaxPacketSize0" = 64
  |     |   "iProduct" = 2
  |     |   "iSerialNumber" = 0
  |     |   "bNumConfigurations" = 1
  |     |   "USB Product Name" = "SE Blank ARIK"
  |     |   "USB Address" = 2
  |     |   "locationID" = 338952192
  |     |   "bDeviceSubClass" = 0
  |     |   "bcdUSB" = 512
  |     |   "Built-In" = No
  |     |   "non-removable" = "no"
  |     |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |     |   "kUSBCurrentConfiguration" = 1
  |     |   "bDeviceProtocol" = 0
  |     |   "USBPortType" = 0
  |     |   "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
  |     |   "USB Vendor Name" = "Freescale SemiConductor Inc "
  |     |   "Device Speed" = 2
  |     |   "idVendor" = 5538
  |     |   "kUSBProductString" = "SE Blank ARIK"
  |     |   "IOGeneralInterest" = "IOCommand is not serializable"
  |     |   "kUSBAddress" = 2
  |     |   "kUSBVendorString" = "Freescale SemiConductor Inc "
  |     |   "IOClassNameOverride" = "IOUSBDevice"
  |     | }
  |     | 
  |     +-o AppleUSBHostLegacyClient  <class IORegistryEntry:IOService:AppleUSBHostLegacyClient, id 0x100000b1d, !registered, !matched, active, busy 0, retain 8>
  |     +-o AppleUSBHostCompositeDevice  <class IORegistryEntry:IOService:AppleUSBHostCompositeDevice, id 0x100000b25, !registered, !matched, active, busy 0, retain 4>
  |     +-o Freescale Flash@0  <class IORegistryEntry:IOService:IOUSBNub:IOUSBInterface, id 0x100000b27, registered, matched, active, busy 0 (23 ms), retain 12>
  |     | +-o AppleUserUSBHostHIDDevice  <class IORegistryEntry:IOService:IOHIDDevice:AppleUserHIDDevice, id 0x100000b29, registered, matched, active, busy 0 (3 ms), retain 13>
  |     |   +-o IOHIDInterface  <class IORegistryEntry:IOService:IOHIDInterface, id 0x100000b4a, registered, matched, active, busy 0 (2 ms), retain 9>
  |     +-o Google Chrome  <class IORegistryEntry:IOService:IOUserClient:IOUSBDeviceUserClient:IOUSBDeviceUserClientV2:IOUSBInterfaceUserClient:IOUSBInterfaceUserClientV2:IOUSBIn$
```

This same devices flashes just fine in Linux. This appears to be due to some driver conflict in MacOS, but it is not clear to me how to further debug this.  Any support is greatly appreciated.

nxpfrankli, 2020-12-28:
Nikolas Grottendieck <git@nikolasgrottendieck.com>  push mac os support.  I have not MAC. 
 Nikolas tested at Catalina.

It because that can't remove mac OS HID driver for IMX devices.

almatthew, 2020-12-28:
> HID driver for IMX devices.

Is there a way to manually remove/disable the macOS HID driver for IMX devices?

nxpfrankli, 2020-12-29:
So far, I have not find a way to do that because need apply key to do that.

Okeanos, 2021-01-18:
The project we were using this for was put on hold in fall last year which is why I now longer have access to the environment. 

However, I think that either restarting or logging out/in again may resolve the issue. I do remember, though, that we experienced issues seemingly unrelated to the hardware/software combination (±1 MacBook Pro HW version and/or macOS version) that we were never able to entirely resolve/reliably reproduce.

roddehugo, 2021-02-17:
Hi,

I'm facing the same issue on macOS Catalina with `libuuu_1.4.43-7-g48764e8`. Sometimes it works, but most often it does not... 

Using `dmesg`, when running `uuu` utility I have:

```
000757.802629 uuu@: AppleUSBHostUserClient::openGated: could not open provider USB3.1 Hub. provider already opened for exclusive access by a kernel client
000757.802969 uuu@: AppleUSBHostUserClient::openGated: could not open provider USB2.1 Hub. provider already opened for exclusive access by a kernel client
000757.803294 uuu@: AppleUSBHostUserClient::openGated: could not open provider USB 3.1 Hub. provider already opened for exclusive access by a kernel client
000757.803412 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.803475 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.803524 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.803571 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.803617 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804019 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804075 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804124 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804171 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804217 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804308 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804354 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804467 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804527 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804577 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804677 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804726 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804785 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804830 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804875 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.804967 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.805012 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.805057 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.805141 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.805185 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.805275 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.805320 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.805365 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.805410 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.805455 uuu@: AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
000757.805765 uuu@: AppleUSBHostUserClient::openGated: could not open provider USB 2.0 Hub. provider already opened for exclusive access by a kernel client
000757.806006 uuu@: AppleUSBHostUserClient::openGated: could not open provider USB 2.0 Hub. provider already opened for exclusive access by a kernel client
000757.806478 uuu@: AppleUSBHostUserClient::openGated: could not open provider USB 10/100/1000 LAN. provider already opened for exclusive access by a kernel client
000757.806794 uuu@: AppleUSBHostUserClient::openGated: could not open provider USB 3.1 Hub. provider already opened for exclusive access by a kernel client
000757.807074 uuu@: AppleUSBHostUserClient::openGated: could not open provider USB 2.0 Hub. provider already opened for exclusive access by a kernel client
000757.807803 uuu@: AppleUSBHostUserClient::openGated: could not open provider USB 2.0 Hub. provider already opened for exclusive access by a kernel client
000757.808017 uuu@: AppleUSBHostUserClient::openGated: could not open provider USB 3.1 Hub. provider already opened for exclusive access by a kernel client
AppleUserUSBHostHIDDevice:0x100000a6b message: 0xe0000101 from: 0x100000a65 0

000758.017069 uuu@: AppleUSBHostUserClient::openGated: could not open provider Freescale Flash. provider already opened for exclusive access by a kernel client
```

When I plug the i.MX6ULL to macOS USB port, I have:
```
Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).
considerRebuildOfPrelinkedKernel prebuild rebuild has expired

AppleUserUSBHostHIDDevice:0x100000ba1 start (state:0x0)

[IOUserUSBHostHIDDevice.cpp:466][0x100000ba1] Open interface: 0x100000b9b
[IOUserUSBHostHIDDevice.cpp:606][0x100000ba1] HID descriptor interface:0 index:0 length:76 76 76
AppleUserUSBHostHIDDevice:0x100000ba1 start (state:0x4)

[... cut ...]
        FBK:                     0x066f  0x9afe
        FBK:                     0x066f  0x9bff
        FB:                      0x0525  0xa4a5
        FB:                      0x18d1  0x0d02
        FB:                      0x3016  0x0001
Wait for Known USB Device Appear...
New USB Device Attached at 20:2
20:2>Start Cmd:SDP: boot -f vsom-6ull-fb.bin
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
20:2>Fail Failure claim interface(0.003s)
```

stu85010, 2021-05-28:
I had same issue, maybe this is related to https://github.com/libusb/libusb/issues/519 ?

roddehugo, 2021-05-28:
FYI, I gave up running the utility under macOS and now rely on a Vagrant machine: 

```
# -*- mode: ruby -*-
# vi: set ft=ruby :

Vagrant.configure("2") do |config|
  config.vm.box = "debian/buster64"
  config.vm.hostname = "buster64.box"
  config.vm.network "private_network", type: "dhcp"

  config.vm.provider "virtualbox" do |vb|
    vb.gui = false
    vb.customize [
      "modifyvm", :id,
      "--name", "sc-mfgtools",
      "--cpus", 1,
      "--memory", "512",
      "--usb", "on",
      "--usbehci", "on",
    ]
    vb.customize [
        'usbfilter',
        'add', '0',
        '--target', :id,
        '--name', 'SE Blank 6ULL',
        '--vendorid', '15a2',
        '--productid', '0080',
    ]
    vb.customize [
        'usbfilter',
        'add', '1',
        '--target', :id,
        '--name', 'USB Gadget',
        '--vendorid', '0525',
        '--productid', 'a4a5',
    ]
  end

  config.vm.provision "shell", env: {"VERSION" => "1.4.72"}, inline: <<-SHELL
    apt-get update
    apt-get upgrade -y
    apt-get install -y \
        build-essential cmake pkg-config \
        libbz2-dev libssl-dev libusb-1.0-0-dev libzip-dev
    wget -q -O /usr/local/bin/uuu \
        https://github.com/NXPmicro/mfgtools/releases/download/uuu_${VERSION}/uuu
    chmod +x /usr/local/bin/uuu
    uuu -udev >> /etc/udev/rules.d/99-uuu.rules
    udevadm control --reload-rules
  SHELL
end
```

And then, from within the same directory as `Vagrantfile`:

```
vagrant ssh -c 'uuu /vagrant/your-uuu-script.uuu'
```

nxpfrankli, 2021-05-28:
Thanks, at least it is one worked solution.

dzhlobo, 2021-08-14:
It seems it was mitigated in upstream libusb: https://github.com/libusb/libusb/pull/911.

I've installed the latest unreleased version of libusb and uuu works well for me: `brew install --head libusb`.

timredfern, 2021-11-06:
The libusb update fixes the "failure claim interface" issue for me in MacOS Mojave, but in Catalina I always get Bulk(W):LIBUSB_ERROR_TIMEOUT. Using a VM always fails