nxp-imx/mfgtools #247: UUU conversion from MFGTool2 imx7d UPT commands

kiwi-ed, 2021-01-19:
Hi,

We are migrating our factory's MFGTool from Windows over to Linux UUU. Converting the ucl2.xml file over to UUU we are stuck at the point when the imx7 jumps/re-enumerates and appears as a UTP device. UUU doesn't appear to support UTP functionality for all the formatting, burning OTP fuses and general Linux commands the ucl2 file had. Is this the right tool for the job?  I've already searched trough the tickets open and closed but have found very little.

Thanks.

nxpfrankli, 2021-01-19:
UUU already don't use UTP.  It change to ufs(FBK) and using android fastboot similar protocol to do communication.

kiwi-ed, 2021-01-19:
Thanks, so what's the way forward? How the fuses etc to be set, format to to UBIFS etc?

nxpfrankli, 2021-01-19:
https://github.com/NXPmicro/mfgtools/wiki/migration-from-ucl2.xml-to-uuu-script

uboot can burn fuse, 

fb: fuse -y xxxx            uboot fuse command format.

kiwi-ed, 2021-01-19:
Appreciate that, great list! So two questions: 1) Our current MFGTool kernel/initramfs/DTB creates a UTP device (virtual USB driver) needed for the second stage being the flashing of the permanent kernel/DTB/FS etc. Can we still us this kernel (and other) or is another one required for FB stage?

2) After loading the initial *.imx,image and rootfs.cpio our product creates a UTP virtual USB drive, the old MFGTool detected this and started loading to it. With UUU after the virtual USB drive is created any other command entered results in "Wait for Known USB Device Appear...". Is this due to using the wrong kernel that creates the virtual USB driver, an incorrect prior SPD command or UUU itself?

Cheers.

nxpfrankli, 2021-01-19:
using latest. https://github.com/NXPmicro/imx-uuc
initramfs should include new linuxrc and ufb, build from ufb.c

And your kernel should enable usb configfs.

kiwi-ed, 2021-01-20:
Arh, that's the key then, makes sense. The documentation that comes with UUU doesn't really mention anything like we've discussed especially the requirement of updated boot kernel/FS having imx-uuc. There are probably others like myself migrating over from mfgtoo2 who look at the ucl2.xml to UUU command sheet and assume it will all - including the mfgtool2 boot kernel/fs - just pull over. 

Thanks again, appreciated you help on this.  Should be fine from here.

closed 2021-01-20 by kiwi-ed