nxp-imx/mfgtools #249: Replacing kernel in with FBK: ucp on i.MX 8MQ

danie007, 2021-02-01:
Hi,

I try to update kernel in **i.MX 8MQ** based SoC with `UUU`. But when I try to do that, I cannot copy the image at the specified location.

<details>
<summary>See UUU error log</summary>
<br>

```
U-Boot SPL 2020.04-5.4.24-2.1.0+g3e4708a941 (Jan 29 2021 - 05:35:14 +0000)
PMIC:  PFUZE100 ID=0x10
DDRINFO: start DRAM init
DDRINFO: DRAM rate 3200MTS
DDRINFO:ddrphy calibration done
DDRINFO: ddrmix config done
Loading 4GB Config
Normal Boot
Trying to boot from USB SDP
SDP: initialize...
SDP: handle requests...
Downloading file of size 923824 to 0x40400000... done
Jumping to header at 0x40400000
Header Tag is not an IMX image
Found header at 0x40407da0
hab fuse not enabled

Authenticate image from DDR location 0x401fcdc0...

U-Boot 2020.04-5.4.24-2.1.0+g3e4708a941 (Jan 29 2021 - 05:35:14 +0000)

CPU:   i.MX8MQ rev2.1 1300 MHz (running at 800 MHz)
CPU:   Industrial temperature grade (-40C to 105C) at 30C
Reset cause: POR
Model: WINSYSTEMS i.MX8MQ ITX-P-C444
DRAM:  4 GiB
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
Loading Environment from MMC... *** Warning - bad CRC, using default environment

[*]-Video Link 0imx8m_hdmi_probe
cdn_api_checkalive: keep-alive counter did not increment for 10us...
HDMI enable failed, ret -1!
 (1280 x 720)
	[0] display-controller@32e00000, video
	[1] hdmi@32c00000, display
In:    serial
Out:   serial
Err:   serial

 BuildInfo:
  - ATF b0a00f2
  - U-Boot 2020.04-5.4.24-2.1.0+g3e4708a941

switch to partitions #0, OK
mmc0(part 0) is current device
Detect USB boot. Will enter fastboot mode!
Net:   Could not get PHY for FEC0: addr 0
Could not get PHY for FEC0: addr 0
No ethernet found.

Fastboot: Normal
Boot from USB for mfgtools
*** Warning - Use default environment for 				 mfgtools
, using default environment

Run bootcmd_mfg: run mfgtool_args;if iminfo ${initrd_addr}; then if test ${tee} = yes; then bootm ${tee_addr} ${initrd_addr} ${fdt_addr}; else booti ${loadaddr} ${initrd_addr} ${fdt_addr}; fi; else echo "Run fas
tboot ..."; fastboot 0; fi;
Hit any key to stop autoboot:  0 

## Checking Image at 43800000 ...
Unknown image format!
Run fastboot ...
Detect USB boot. Will enter fastboot mode!
flash target is MMC:0
request 00000000fe130200 was not queued to ep1in-bulk
Detect USB boot. Will enter fastboot mode!
request 00000000fe130200 was not queued to ep1in-bulk
switch to partitions #0, OK
mmc0(part 0) is current device
Detect USB boot. Will enter fastboot mode!
request 00000000fe130200 was not queued to ep1in-bulk
```

</details>

I tried adding `CONFIG_USB_F_FS=y` following this issue: https://github.com/NXPmicro/mfgtools/issues/42#issuecomment-408021687. Yet it's not working as expected.

The script I used is given as follows,
```bash
uuu_version 1.2.39

SDP: boot -f >/signed_flash.bin

SDPS: boot -f >/signed_flash.bin

SDPV: delay 1000
SDPV: write -f >/signed_flash.bin -skipspl
SDPV: jump

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev 0
FB: ucmd mmc dev 0

FBK: ucp t:/boot/Image file.txt

FB: Done

```

nxpfrankli, 2021-02-01:
You need download linux kernel image and initramfs before using FBK. 

our release package have example_kernel_emmc.uuu.  

https://github.com/NXPmicro/mfgtools/wiki/Sample-script

section: 
Burn yocto image to eMMC by linux kernel

danie007, 2021-02-03:
Thanks @nxpfrankli 

Do you have any idea on generating `.initramfs` file in Yocto?

nxpfrankli, 2021-02-11:
I have not built it by myself. 
looking for our bsp yocto document.
try bitbake core-image-minimal-initramfs

closed 2021-02-11 by nxpfrankli