nxp-imx/mfgtools #250: Not able to flash iMX8QX MEK using UUU

ramprakashb, 2021-02-04:
Hi All,

I am trying to boot QSPI flash memory in iMX8QX MEK evaluation board using latest UUU. I have used both custom built image for QSPI and also the pre built image "imx-boot-imx8qxpmek-fspi.bin-flash_flexspi" from the NXP website. But whenever I use the command "sudo uuu -b qspi flash_uboot.bin flash_scfw.bin " I get the following error. How to fix this issue?

nikola@NikolaTesla:iMX8QX$ sudo uuu -v -b qspi flash_uboot.bin flash_scfw.bin 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.154-0-g47ddef5

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02

Run built-in script:
 
uuu_version 1.2.39

# @_flexspi.bin            | bootloader
# @_image   [_flexspi.bin] | image burn to flexspi, default is the same as bootloader

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f _flexspi.bin

# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM, skip QSPI header
SDPS: boot -f _flexspi.bin -skipfhdr

# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
# {
SDPU: delay 1000
SDPU: write -f _flexspi.bin -offset 0x10000 -skipfhdr
SDPU: jump
# }

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f _flexspi.bin -skipspl -skipfhdr
SDPV: jump
# }

FB: ucmd setenv fastboot_buffer ${loadaddr}
FB: download -f _image

# Check Image if include flexspi header
FB: ucmd if qspihdr dump ${fastboot_buffer}; then setenv qspihdr_exist yes; else setenv qspihdr_exist no; fi;

FB[-t 60000]: ucmd if test ${qspihdr_exist} = yes; then qspihdr init ${fastboot_buffer} ${fastboot_bytes} safe; else true; fi;

#if uboot can't support qspihdr command, use uboot image to write qspi image, which require image include qspi flash header
FB: ucmd if test ${qspihdr_exist} = no; then sf probe; else true; fi;
FB[-t 40000]: ucmd if test ${qspihdr_exist} = no; then sf erase 0 +${fastboot_bytes}; else true; fi;
FB[-t 20000]: ucmd if test ${qspihdr_exist} = no; then sf write ${fastboot_buffer} 0 ${fastboot_bytes}; else true; fi;
FB: done

Wait for Known USB Device Appear...

Error: fail open file: >_flexspi.bin

nikola@NikolaTesla:iMX8QX$ 

Thanks and Regards,
Ramprakash
[image]
[image]

nxpfrankli, 2021-02-04:
can you try latest uuu?
1.3.154 is old one.

it is strange.  all _file should replace by followed argument.

ramprakashb, 2021-02-05:
Hi @nxpfrankli ,

Thanks for the reply.

I did try with the latest version. But again for the latest demo images I get the following error:

[image]
[image]
[image]
[image]

nxpfrankli, 2021-02-07:
are you C0 chip or B0 chip?

ramprakashb, 2021-02-08:
Hi @nxpfrankli ,

I am using a C0 chip.

Thanks and Regards,
Ramprakash

nxpfrankli, 2021-02-08:
you should use c0 file

imx-boot-imx8qxp**c0**mek-fspi.bin-flash_spl_flexspi

ramprakashb, 2021-02-08:
Hi @nxpfrankli ,

Thank you the demo boot image imx-boot-imx8qxpc0mek-fspi.bin-flash_spl_flexspi worked and the board is booted with u-boot. 

But when I try to use UUU for the custom image built using mkimage tool. I still get the same error. Is there any changes that are required for building the images for C0 board.

[image]
[image]

nxpfrankli, 2021-02-08:
Added REV=C0

ramprakashb, 2021-02-08:
I tried the command "make SOC=iMX8QXP REV=C0 " but I face error like the below. 

I copied the files from the location "cp -r platform/board/mx8qx_val platform/board/mx8qx_myboard" before executing the make command.
[image]

nxpfrankli, 2021-02-09:
This command is for imx_mkimage, not build scfw.

Imx_mkimage package SECO firmware contrainer, uboot, scfw together and build one image, which rom can recognize.

Caution: EXT Email

[Screen Shot 2021-02-08 at 3 22 59 PM]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F41544795%2F107294323-0e859580-6a22-11eb-9101-4c7320883456.png&data=04%7C01%7Cfrank.li%40nxp.com%7C8be68a41a1da41c7458508d8cc890d5f%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C637484236297380265%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=a64FYyzG1enuG2tT0VwVlrxgL%2FduImcCJ3Eq%2FlD%2BfTo%3D&reserved=0>

-

ramprakashb, 2021-02-09:
Hi @nxpfrankli ,
Thanks for the reply.

I did try the REV=C0 with mkimage tool. But when I use the REV = C0 the command which gets executed still shows B0. And the image fails to flash SPI memory.

But when I try to hardcode the REV=C0 into the soc.mak file under iMX8QX folder I get a error that the Rev = C0 is not known.  Do you have any idea how to fix this?
[image]
[image]

nxpfrankli, 2021-02-09:
Did your mkimage too old

Caution: EXT Email

[Screen Shot 2021-02-08 at 5 42 54 PM]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F41544795%2F107304275-9d4fdd80-6a35-11eb-9a51-e25255176746.png&data=04%7C01%7Cfrank.li%40nxp.com%7Cdb4aed7fdc0d499c47b508d8cc9c9a9c%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C637484320262066949%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=ecJBkrTQIgoxcmdimixc5kCnTDQWVzYmH60ZGZN3BW0%3D&reserved=0>
[Screen Shot 2021-02-08 at 5 44 21 PM]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F41544795%2F107304278-9de87400-6a35-11eb-8530-1081a647bde8.png&data=04%7C01%7Cfrank.li%40nxp.com%7Cdb4aed7fdc0d499c47b508d8cc9c9a9c%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C637484320262066949%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=03qbShGJaN4R2EusJo5MifjD2bD%2Bw6c9%2FWNIeKNBiHM%3D&reserved=0>

-

ramprakashb, 2021-02-10:
Hi @nxpfrankli ,

Thanks for the reply.

I now got the latest mkimage from the NXP website and the when I try to use uuu for the flash_flexspi image which I built, the output reaches 100% and then gets suck there. It does not moves past it. I did not face this issue while flashing demo image. Is the command execution correct or do I need to add any other file together with this?

[image]
[image]

nxpfrankli, 2021-02-10:
your image have not boot in customer board. 
Please check your image.

ramprakashb, 2021-02-10:
Hi @nxpfrankli ,

For imx8 C0 version of the board apart from SECO is there any changed required in the build process.

I have generated the image using the following components:
SCU - 
- Made a copy of imx8qx_val as myboard and then built the image using the command:
make qx R=B0 B=myboard
- I tried using C0 here but it throws error for REV=C0 

SECO:
- I got the SECO binary for C0 from the NXP website.
- The version number is L4.14.98_2.0.X_SCFWKIT-1.2.11

U-Boot:
- I secured the package from U-boot website
- I don't see anything in particular for C0 board.
make imx8qxp_mek_defconfig
- I used the above command to generate the configs and used make to build the u-boot binary file.

ATF:
- Here again I did not see anything separate for c0 board.
- I just used the command 
make PLAT=imx8qx bl31
- This generated the bl31 binary file.

Mkimage:
- I used all the above mentioned files in the latest version of mkimage
- And generated the flash image using the command:
make REV=C0 SOC=iMX8QX flash_flexspi
- I renamed the flash image and used the UUU tool to boot the SPI flash
sudo uuu -v -b qspi iMX8QX/flash.bin

Is there anything missing in the above steps?