nxp-imx/mfgtools #251: FB[-t timeout]: doesn't work in the command line

gomme600, 2021-02-10:
Hi,

If I use the following command in a uuu .conf script it works fine: "FB[-t 10000]: ucmd sf erase 0x00000 0xc0000"
But running the following from the command line returns an error: "uuu -m 1:10 FB[-t 10000]: ucmd sf erase 0x00000 0xc0000" returns "Error: fail open file: >FB[-t"

This behavior seems like a bug and breaks my usage of uuu with LAVA as lava doesn't execute a .conf file but rather a succession of commands.

Thanks for any help

gomme600, 2021-02-10:
Ok, I have just found a workaround for anyone looking, here it is: uuu FB['-t 10000']: ucmd sf erase 0x00000 0xc0000

gomme600, 2021-02-10:
Ok, so this doesn't work in LAVA for some reason, I think that when LAVA applies the command then the single quotes get ignored so I still think that this is a uuu bug as the single quotes shouldn't be needed in the first place.

gomme600, 2021-02-10:
A workaround seems to be using FB[-t]:, I think that this must set the timeout to unlimited but without using a space in the brackets.

nxpfrankli, 2021-02-10:
use double quarto

uuu -m 1:10 "FB[-t 10000]:" ucmd sf erase 0x00000 0xc0000

closed 2021-02-11 by nxpfrankli