nxp-imx/mfgtools #252: libusb submodule version is not exist

tonyho, 2021-02-11:
Checkout Error:

```
 git submodule update --init --recursive
Submodule 'bzip2' (git://sourceware.org/git/bzip2.git) registered for path 'bzip2'
Submodule 'libusb' (https://github.com/nxpfrankli/libusb.git) registered for path 'libusb'
Submodule 'zlib' (https://github.com/madler/zlib.git) registered for path 'zlib'
Cloning into 'bzip2'...
remote: Enumerating objects: 473, done.
remote: Counting objects: 100% (473/473), done.
remote: Compressing objects: 100% (247/247), done.
remote: Total 473 (delta 344), reused 304 (delta 225), pack-reused 0
Receiving objects: 100% (473/473), 598.06 KiB | 441.00 KiB/s, done.
Resolving deltas: 100% (344/344), done.
Checking connectivity... done.
Submodule path 'bzip2': checked out '6a8690fc8d26c815e798c588f796eabe9d684cf0'
Cloning into 'libusb'...
remote: Enumerating objects: 13705, done.
remote: Total 13705 (delta 0), reused 0 (delta 0), pack-reused 13705
Receiving objects: 100% (13705/13705), 3.80 MiB | 1.52 MiB/s, done.
Resolving deltas: 100% (9945/9945), done.
Checking connectivity... done.
fatal: reference is not a tree: e782eeb2514266f6738e242cdcb18e3ae1ed06fa
Cloning into 'zlib'...
remote: Enumerating objects: 5150, done.
remote: Total 5150 (delta 0), reused 0 (delta 0), pack-reused 5150
Receiving objects: 100% (5150/5150), 3.66 MiB | 1.29 MiB/s, done.
Resolving deltas: 100% (3612/3612), done.
Checking connectivity... done.
Submodule path 'zlib': checked out 'cacf7f1d4e3d44d871b605da3b647f07d718623f'
Unable to checkout 'e782eeb2514266f6738e242cdcb18e3ae1ed06fa' in submodule path 'libusb'

```

nxpfrankli, 2021-02-11:
strange, https://github.com/nxpfrankli/libusb/tree/e782eeb2514266f6738e242cdcb18e3ae1ed06fa
and ci have not report problem

Lin-Buo-Ren, 2021-03-06:
@tonyho 

It appears to be [a deficiency/bug in Git <2.8 that makes Git unable to check-out the submodule hash](https://stackoverflow.com/a/17692710/6219078):

git 2.7.4(ubuntu 16.04) - failure
git 2.17.1 (ubuntu 18.04) - success
git 2.25.1 (ubuntu 20.04) - success
git 2.27.0 - success

Possible solution:

* Upgrade Git before checking out submodule
* Migrate to a recent Ubuntu Core base(`core18`) that ships a recent Git distribution

closed 2021-03-10 by nxpfrankli