nxp-imx/mfgtools #254: hang after loading uboot in a docker container

kuhyeonjeong, 2021-02-25:
I'm trying to flash an Android device in a docker container. Normally, uuu works fine in my host Ubuntu but when I run uuu in a docker container(Ubuntu), it hangs right after loading uboot image. So when it hung, I tried to kill the process and ran uuu with my command list, again. Then, it went through the rest of commands.
Is it possible to run uuu in docker container?

```uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.95-0-g38dc325

Build in config:
        Pctl     Chip            Vid     Pid     BcdVersion
        ==================================================
        SDPS:    MX8QXP          0x1fc9  0x012f  [0x0002..0xffff]
        SDPS:    MX8QM           0x1fc9  0x0129  [0x0002..0xffff]
        SDPS:    MX8DXL          0x1fc9  0x0147
        SDPS:    MX28            0x15a2  0x004f
        SDPS:    MX815           0x1fc9  0x013e
        SDPS:    MX865           0x1fc9  0x0146
        SDPS:    MX8ULP          0x1fc9  0x014a
        SDPS:    MX8ULP          0x1fc9  0x014b
        SDP:     MX7D            0x15a2  0x0076
        SDP:     MX6Q            0x15a2  0x0054
        SDP:     MX6D            0x15a2  0x0061
        SDP:     MX6SL           0x15a2  0x0063
        SDP:     MX6SX           0x15a2  0x0071
        SDP:     MX6UL           0x15a2  0x007d
        SDP:     MX6ULL          0x15a2  0x0080
        SDP:     MX6SLL          0x1fc9  0x0128
        SDP:     MX7ULP          0x1fc9  0x0126
        SDP:     MXRT106X        0x1fc9  0x0135
        SDP:     MX8MM           0x1fc9  0x0134
        SDP:     MX8MQ           0x1fc9  0x012b
        SDPU:    SPL             0x0525  0xb4a4  [0x0000..0x04ff]
        SDPV:    SPL1            0x0525  0xb4a4  [0x0500..0x9998]
        SDPU:    SPL             0x0525  0xb4a4  [0x9999..0x9999]
        SDPU:    SPL             0x3016  0x1001  [0x0000..0x04ff]
        SDPV:    SPL1            0x3016  0x1001  [0x0500..0x9998]
        FBK:                     0x066f  0x9afe
        FBK:                     0x066f  0x9bff
        FB:                      0x0525  0xa4a5
        FB:                      0x18d1  0x0d02
        FB:                      0x3016  0x0001
Wait for Known USB Device Appear... at path 1:1 
New USB Device Attached at 1:1
1:1>Start Cmd:SDPS: boot -f u-boot-imx8qxp-xxxxxxxxx.imx
100%1:1>Okay (2.331s)
```

nxpfrankli, 2021-02-25:
You need config docker, 

Pass below USB VID and PID to docker.  

        SDPV:    SPL1            0x0525  0xb4a4  [0x0500..0x9998]
        SDPU:    SPL             0x0525  0xb4a4  [0x9999..0x9999]
        SDPU:    SPL             0x3016  0x1001  [0x0000..0x04ff]
        SDPV:    SPL1            0x3016  0x1001  [0x0500..0x9998]
        FBK:                     0x066f  0x9afe
        FBK:                     0x066f  0x9bff
        FB:                      0x0525  0xa4a5
        FB:                      0x18d1  0x0d02
        FB:                      0x3016  0x0001

kuhyeonjeong, 2021-02-25:
> You need config docker,
>
> Pass below USB VID and PID to docker.
>
> ```
>     SDPV:    SPL1            0x0525  0xb4a4  [0x0500..0x9998]
>     SDPU:    SPL             0x0525  0xb4a4  [0x9999..0x9999] [...]

@nxpfrankli Thank you very much for the reply. My docker has configured as privileged mode and it mounts entire /dev as volume so I think all USBs are accessible inside the container just like my host.
I’m so sorry that I’m new to this tool so please correct me if I’m wrong. Is there something I need to config my container?

nxpfrankli, 2021-02-25:
I think udev have not notified uuu. 
can you uuu again after sdps?
or run lsusb, make sure usb device  existed.

kuhyeonjeong, 2021-02-25:
@nxpfrankli As you said, it seems udev issue. Using host network mode for the docker container resolved the issue. Thanks for your help!

closed 2021-03-05 by nxpfrankli