nxp-imx/mfgtools #264: emmc_all command not working properly

israelzimmerman, 2021-03-15:
The sequence for emmc_all burn the full image and after that it is trying to flash the uboot used for uuu
also the mmc partconf command does not un-protect  the flash
The fix was checked and attached her on the file uuu/emmc_burn_all.lst

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${emmc_dev}
FB: ucmd mmc dev ${emmc_dev}
FB: ucmd if env exists emmc_ack; then ; else setenv emmc_ack 1; fi;
FB: ucmd mmc partconf ${emmc_dev} ${emmc_ack} 0 0
FB: flash -raw2sparse all _rootfs.sdcard
FB: done

nxpfrankli, 2021-03-16:
Please send out pull request

nxpfrankli, 2021-07-28:
original partconf is correct. 
FB: ucmd mmc partconf ${emmc_dev} ${emmc_ack} 1 0

```
mmc partconf

EXT_CSD[179], PARTITION_CONFIG:
BOOT_ACK: 0x0
BOOT_PARTITION_ENABLE: 0x1
PARTITION_ACCESS: 0x0
```

FB: ucmd mmc partconf ${emmc_dev} ${emmc_ack} 0 0
mmc partconf dev
```
EXT_CSD[179], PARTITION_CONFIG:
BOOT_ACK: 0x0
BOOT_PARTITION_ENABLE: 0x0
PARTITION_ACCESS: 0x0
```

flash bootloader is necessary part because bootloader write to boot partition.  _rootfs.sdcard may include bootloader, but  it is located in user partition.

So user just need burn fuse to enable emmc fastboot.

Some rootfs is common for multi-platform,  need use correct uboot to overwrite what saved in rootfs.

nxpfrankli, 2021-12-02:
no feedback for long time, if still problem, please reopen

closed 2021-12-02 by nxpfrankli