nxp-imx/mfgtools #269: snap post refresh hook not work

nxpfrankli, 2021-04-23:
@Lin-Buo-Ren:

I try to add udev rule in post refresh hook. but it doesn't work.  
Can you help check?

Lin-Buo-Ren, 2021-04-23:
As far as I can tell currently it is not possible to modify udev rules from with-in the snap, the root filesystem (/) from the snapped applications' perspective is actually a Ubuntu filesystem provided by the `core*` snap instead of the actual host filesystem thus placing udev drop-in rules under /etc/udev/rules.d will not have any intended effect.

For hardware integration support one must rely on the facilities snapd provided, which is currently limited to the usage of USB serial interface devices under Ubuntu Core (not _classic_) systems, refer the following references for more info:

* [Hotplug support | Snapcraft documentation](https://snapcraft.io/docs/hotplug-support)
* [The serial-port interface | Snapcraft documentation](https://snapcraft.io/docs/serial-port-interface)
* [Developing hotplug interfaces | Snapcraft documentation](https://snapcraft.io/docs/developing-hotplug-interfaces)
* [How to add or workaround a udev rule - snapd - snapcraft.io](https://forum.snapcraft.io/t/how-to-add-or-workaround-a-udev-rule/12829)
* The Device access section in [Multiple users and groups in snaps - snapd - snapcraft.io](https://forum.snapcraft.io/t/multiple-users-and-groups-in-snaps/1461)

Currently the only way to achieve UUU's typical usage(connecting & flashing USB devices) is to run it as root and thus have the permission to write to the devices (via the already granted [raw-usb interface](https://snapcraft.io/docs/raw-usb-interface), it is snapd's limitation yet to be lifted.

Lin-Buo-Ren, 2021-04-23:
Another possibility is to instruct the user to place udev rules themselves in the host system's /etc/udev/rules.d, which is already done by UUU's `-udev` command-line option IIRC.