nxp-imx/mfgtools #270: Relation between UUU2 flashing and Windows OS performance

ajithpv, 2021-05-04:
Hi All,

We are using UUU.exe (UUU version(libuuu) 1.3.102) on Windows 10 OS for flashing i.MX8 board.

We came across a strange issue as described below:

- The UUU is working fine as long as windows 10 performance is fine (say 9 out of 10 flashing is OK). However, during the continuous flashing (for 1 month to 2 month time frame), the Windows 10 performance is degrading (even opening the Windows file explorer  is slow). During this degradation, UUU2 flashing also failing at very high rate (say 8 out of 10 flashing are failing).
- In all the error case, we are getting the below error:
```
New USB Device Attached at 2:12
2:12>Fail Failure open usb device(1.845e+16s)
```
- To confirm this , we replace the Windows 10 with a fresh back-up copy and done the flashing. Then the flashing again become good (say 9 out of 10 flashing is OK ).
- We are not getting why Windows 10 performance degradation affect the UUU libusb flashing or how it is related.

Could anyone please let me know why this is happening? 
Also, any clue on how to resolve this issue is appreciated.

nxpfrankli, 2021-05-04:
It may be caused by registry. 

ref
https://www.microchip.com/forums/m543480.aspx

https://www.ftdichip.com/Support/Knowledgebase/index.html?ignorehardwareserialnumber.htm
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-device-specific-registry-settings

Add IgnoreHWSerNum entry under this key,
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\UsbFlags\

The entry name should have VID/PID of MCP2200 (04D8/00DF) like,
IgnoreHWSerNum04D800DF

Set its value to 0x01 (REG_BINARY, length must be exactly one byte)

IgnoreHWSerNum is helpful for production and testing of USB devices.

ajithpv, 2021-05-06:
Thank you for your reply.
I would like to inform you that we are flashing 2 boards at the same time using UUU (using -m option to specify the port number). 
Could you please let me know whether the above change have any impact on the parallel flashing?

nxpfrankli, 2021-05-06:
no.

closed 2021-05-20 by ajithpv

rajumallah, 2021-09-29:
Hi All,

Can you please provide a list of VID used for Ublox which need to be added in ignred list because we are seeing in registry that there are several other VIDs are registered. Since with every device getting connected to PC registry is growing and it is causing system degradation. 

Generally size of registry is around 400MB for production pc, but after production of several hundreds of devices connected to pc, it size grown upto 1.5 GB.

Thanks,
Raju Mallah