nxp-imx/mfgtools #271: flash time

rschwarz-sotec, 2021-05-27:
Hi

On a custom imx7 board we try to flash with uuu an 8GB sdimg. command:
`FB: flash -raw2sparse all IMAGE.sdimg`
This takes about 10 minutes.

Is there a way to speed up flashing? What is the recommended way of doing this? How to flash thousands of devices in a manufacturing process?
We want to keep manufacturing process as simple as possible and therefore want to avoid starting linux from ramdisk with partitioning, flashing, ...

Thanks in advance 
Ruben

nxpfrankli, 2021-05-27:
1. Try build one small sdimg.  
2. I am not sure if there are good tools can convert to android sparse file format, which can discard some data, which file system don't care. 
3. you connect many devices into difference port, so many devices can be programmed parallel.
4. Remove message from uart port.  Debug message takes some time to transfer to uart port even no connected.

rschwarz-sotec, 2021-06-17:
Thanks @nxpfrankli for your response and sorry for late response.

regarding point 3 difference ports: I have to burn per-device data like mac-address and serial number. The values are provided by a script which is called from uuu with stdout (see [link](https://github.com/NXPmicro/mfgtools/wiki/UUU-default-support-protocol-list#common-command-for-all-protocol).
If I have multiple information (e.g. mac address needs 3 fuses to burn), how can I handle concurrency here? Is there a identifier from uuu which I can use as parameter for my script which provides mac addresses?

nxpfrankli, 2021-06-17:
So far, no. maybe you need a global counter file by shared file.  I leave flexibility to user.

rschwarz-sotec, 2021-06-17:
Do you have any idea how to share information between 2 calls from uuu script?
Otherwise i have to split into two steps: one long-running parallel flash job and a second, fast configure job which runs sequential

nxpfrankli, 2021-06-17:
```
counter(){
    a=$1
    while true;do
        if [ ! -e /tmp/counter.lock ];then
            touch /tmp/counter.lock
            curr_count=`cat /tmp/count.txt`
            n_count=`echo "${curr_count}  + $a" | bc`
            echo ${n_count} > /tmp/count.txt
            rm -fv /tmp/counter.lock
            break
        fi
        sleep 1 
    done
}
```

closed 2021-07-28 by nxpfrankli