nxp-imx/mfgtools #272: ERROR - "Back to ROM, SDP" on i.MX 8M Mini

danie007, 2021-06-02:
We're trying to flash the bootloader on [CompuLab IOT-GATE-iMX8 EVK](https://www.compulab.com/products/iot-gateways/iot-gate-imx8-industrial-arm-iot-gateway/#evalkit) that is based on **NXP i.MX 8MM application processor**, we get the following error.

```shell
U-Boot SPL 2020.04-iot-gate-imx8-2.3-5.4.24-2.1.0+g4979a99482 (May 28 2021 - 06:32:44 +0000)
power_bd71837_init
DDRINFO(D): Nanya 2048G
DDRINFO: start DRAM init
DDRINFO:ddrphy calibration done
DDRINFO: ddrmix config done
DDRINFO(M): mr5-8 [ 0x5000010 ]
DDRINFO(E): mr5-8 [ 0x5000010 ]
Back to ROM, SDP
```

We referenced [this issue](https://github.com/NXPmicro/mfgtools/issues/93#issuecomment-480384075), which talks about "Back to ROM, SDP" error and verified that we're using the latest and supported U-Boot version (v2020.04).

> It is to be noted that, the above error message gets printed as soon as we changed to SDP. And when we try to run the `UUU` flashing script, it does not respond at all.

**NOTE**: We checked that the **SDP** is working by running the `uuu -lsusb` command and also checked that `fastboot` is enabled and working on the U-Boot.

---

CC @tkmozhi

danie007, 2021-06-02:
@nxpfrankli can you comment on this?

nxpfrankli, 2021-06-02:
Back to ROM, SDP is old method. 
We have give up this way. 

Please check config and nxp release uboot. 
nxp 2020 uboot should already switch SPL SDP.

danie007, 2021-06-03:
Our U-Boot version is 2020.04 yet we're seeing this error! What does that signify?

> Please check config and nxp release uboot.

What should I be checking on these? Could you be more specific.

---

CC @tkmozhi

nxpfrankli, 2021-06-03:
It is boards file.  refer: board/freescale/imx8mm_evk/spl.c

int spl_board_boot_device(enum boot_device boot_dev_spl)
{
        switch (boot_dev_spl) {
        case SD2_BOOT:
        case MMC2_BOOT:
                return BOOT_DEVICE_MMC1;
        case SD3_BOOT:
        case MMC3_BOOT:
                return BOOT_DEVICE_MMC2;
        case QSPI_BOOT:
                return BOOT_DEVICE_NOR;
        case NAND_BOOT:
                return BOOT_DEVICE_NAND;
        case USB_BOOT:
                return **BOOT_DEVICE_BOARD**;
        default:
                return BOOT_DEVICE_NONE;
        }
}

Need return BOOT_DEVICE_BOARD.

danie007, 2021-06-04:
@nxpfrankli I've verified the above functions on the NXP i.MX 8MM EVK and everything works as expected. However in [CompuLab BSP](https://github.com/compulab-yokneam/meta-bsp-imx8mm/tree/iot-gate-imx8_r2.4), it's seem like that they've scraped the SDP related parts.

If the CompuLab BSP is void of these options, **how should I add these**? Are there any references?

Thanks in advance.

---

CC @tkmozhi

nxpfrankli, 2021-06-04:
what's your means?  board/freescale/imx8mm_evk/spl.c is refer code.

closed 2021-07-28 by nxpfrankli

danie007, 2021-07-29:
@nxpfrankli Thanks for your support 😇
The issue is resolved with this [patch note](https://github.com/compulab-yokneam/meta-bsp-imx8mm/commit/a804dc61c80f6193a1a8a1f9914c89b31ff7f031#commitcomment-52128021) from one of the collaborators (@vraevsky) of [compulab-yokneam/meta-bsp-imx8mm](https://github.com/compulab-yokneam/meta-bsp-imx8mm/tree/iot-gate-imx8-dev) repo.

---

@tkmozhi