nxp-imx/mfgtools #276: iMX6 DL

PREMjITHT, 2021-07-24:
First of all , I hope I have posted my query in the right place . I am expecting software support/suggestions for the below query.

I am using MFG Tool to flash iMX6 DL board.
Now I am trying to migrate it from MFG Tool to UUU tool.
Is there is any dependency for UUU Tool ? Whether it supports iMX6DL ?

nxpfrankli, 2021-07-26:
Latest NXP released BSP already support uuu for iMX6 DL.
If you stick old version uboot\kernel, you need check uboot and kernel configuration requirement. 
and update imx-uuc for your initramfs.  or you directly download latest prebuild initramfs.

PREMjITHT, 2021-07-26:
Hi @nxpfrankli 

As you can see in the [bootlog.txt](https://github.com/NXPmicro/mfgtools/files/6880650/bootlog.txt)  file, Uboot version is 2019.04 which is supported by uuu tool.

I am not able to boot the device, **acmd booti** is not working for me. 
Getting below error while runing the uuu tool.
**Unknown command 'booti' - try 'help'**

This is the uuu script -  [script.txt](https://github.com/NXPmicro/mfgtools/files/6880636/script.txt)

nxpfrankli, 2021-07-26:
your uboot have built in booti command. check uboot config CONFIG_CMD_BOOTI

closed 2021-07-28 by nxpfrankli